PX4/PX4-SITL_gazebo-classic #317: controlling typhoon_480 gimbal (via mavros - /mavros/actuator_control message)

jannsta1, 2019-05-08:
Have tried to do this a few times but never seem to be able to get it working well so wondering if there are any examples of controlling the typhoon gimbal via the /mavros/actuator_control message?

If I set the actuator_control message group_mix to 2 and ```controls``` to a length 8 array, the first few elements seem to correspond to roll, pitch, yaw channels respectively but the gimbal responds in an extremely jittery fashion, enough to also effect the simulated drone's position control stability. 

While sending the above message, if I run ```gz topic -e /gazebo/default/typhoon_h480/gimbal_yaw_cmd``` the setpoint oscillates between -2.19 and 0 whereas the rostopic (/mavros/actuator_control) remains stable. I guess its a problem in the logic of the gazebo_gimbal_controller_plugin.cpp file(?) but struggling to identify exactly what the issue is.

RicardoM17, 2019-05-08:
Hello @jannsta1 

I spent quite some time this week in this subject so I'm also very interested in hearing if there's a better way or where exactly is the problem.

Like you I also have a node publishing via the /mavros/actuator_control message, group_mix 2 using the controls. I also experience very weird behaviour with the gimbal, completely unstable and jittery. My next step was to actually modify the sdf file and change the pid values of the gimbal and though I still have to play with it a bit more it didn't seem to help. Iirc from this example: https://dev.px4.io/en/advanced/gimbal_control.html#testing the performance is completely different so it seems like there is an issue with the gimbal plugin like you mentioned.

I also get some other issues with gazebo where the real time factor decreases a lot for certain periods of time causing the drone to flip around and almost crash. I was just wondering if you ever had that issue too but I'm not sure if it's related. I'm running a modified version of the offboard example.

jannsta1, 2019-05-08:
Hi @RicardoM17,

I'm pretty sure that the problem is at least partly to do with conflicting gimbal control (input) messages. I'm away from my computer for a day or two so unfortunately can't check the real time factor at the moment, will do when I get back. A few things I can mention:

- with the example you mention, it says "if you send mavlink commands, set the stabilise flags to false". I think this is referring to the flags in the MAV_CMD_DO_MOUNT_CONFIGURE message(?) as described in https://mavlink.io/en/messages/common.html. Since mavros is using the SET_ACTUATOR_CONTROL_TARGET message I don't think this applies, or at least I don't think there is  a way to set these flags from mavros without creating a new plugin. Maybe I'm misreading this though? 
- I tried looking for message conflicts between MAV_CMD_DO_MOUNT_CONFIGURE and SET_ACTUATOR_CONTROL_TARGET in the QGroundControl mavlink widget but couldn't see either message, even though the second one must be being sent - perhaps there is a better way to view mavlink messages?
- The vmount module in px4 could be the source of the conflicting control message. I thought I'd tried deactivating this but I could be wrong.
-  maybe there is some logic error in gazebo_gimbal_controller_plugin.cpp. My next step will probably be to uncomment some of the gzdbg lines for one of the channels in order to get an idea what's happening here.

jannsta1, 2019-05-10:
Quick update - I checked disabling vmount and this stopped the conflicting setpoint message (/gazebo/default/typhoon_h480/gimbal_yaw_cmd now reflects actuator_control message all the time (post arming)) so that the gimbal will yaw to the desired setpoint. Pitch and roll commands don't seem to be actioned by the gimbal model as far as I can see but the gimbal_pitch/roll_cmd messages is updated as expected

However, the uav becomes unstable whenever the actuator controls message is sent, even if all values setpoint values are sent to 0 (i.e. the weight distribution and control actions should be the same as when this message is not sent). Any suggestions on what could be causing this?

Jaeyoung-Lim, 2019-05-15:
@jannsta1 Have you tried lowering the gimbal setpoint rates?

jannsta1, 2019-05-15:
Thanks @Jaeyoung-Lim - I changed the ROS ActuatorControl message publish rate from 10hz to 1hz which appears to have largely fixed the issue. 

In this instance this solution is OK for me but I wonder if this is something that could be made more robust for others. Do you know anything about where this problem arises? (I'm not sure if its a mavros thing, Px4 firmware or perhaps its just associated with gazebo simulator - do you know if this LED problem was on real hardware or also in (gazebo) simulation?)

RicardoM17, 2019-05-15:
@Jaeyoung-Lim @jannsta1 Once I have time I will also test this but it's looking like we found the issue and at least for now a quick fix. I quickly tried on my example but like I mentioned I'm having some issues with the Real Time Factor in Gazebo. If I stop the publishing of the gimbal controls then this issue stops so maybe it's related.
Later I will try to separate the publishing of the two messages so that they go in different rates.
@jannsta1 Would you be Ok with sharing the part of your code related to this? I am curious to see your implementation.

jannsta1, 2019-05-15:
Python code for a test actuator controls ROS node included below. Sorry, forgot to check on the realtime factor, I don't remember seeing a high number of ```[timesync] Time jump detected.``` messages but will try and check this next time I run it.

```
import rospy
import numpy as np
from mavros_msgs.msg import ActuatorControl
from geometry_msgs.msg import PoseStamped
from tf.transformations import euler_from_quaternion

class Gimbal_test():

    def __init__(self,
                 ):

        self.message_pub = rospy.Publisher("/mavros/actuator_control", ActuatorControl, queue_size=10)
        self.local_pos_sub = rospy.Subscriber('mavros/local_position/pose',
                                              PoseStamped,
                                              self.local_position_callback)

        self.actuator_control_message = ActuatorControl()
        self.seq = 0
        self.local_position = PoseStamped()
        self.heading = 0.0

    def local_position_callback(self, data):
        self.local_position = data
        self.heading = self.quat2yaw(data.pose.orientation)

    def quat2yaw(self, this_quat):
        '''
        converts from a quaternion (developed using message type: posestamped.pose.orientation) to a yaw
        we assume that the UAV is level enough for this to work
        :param this_quat: 
        :return: 
        '''
        (_, _, yaw) = euler_from_quaternion([this_quat.x, this_quat.y, this_quat.z, this_quat.w])
        return yaw

    def run(self):

        r =rospy.Rate(1)
                        #   r    p    y?
        inputs = np.array((0.0, 0.3, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0))

        while not rospy.is_shutdown():

            self.actuator_control_message.header.stamp = rospy.Time.now()
            self.actuator_control_message.header.seq = self.seq
            self.actuator_control_message.group_mix = 2 # ActuatorControl.PX4_MIX_PAYLOAD
            rospy.loginfo_throttle(3, inputs)
            self.actuator_control_message.controls = inputs
            self.message_pub.publish(self.actuator_control_message)

            self.seq = self.seq + 1
            r.sleep()

if __name__ == '__main__':

    rospy.init_node('gimbal_test', anonymous=True, log_level= rospy.INFO)
    gt = Gimbal_test()
    gt.run()
```

Jaeyoung-Lim, 2019-05-16:
@jannsta1 I think the core issue is might be related to the firmware being able to only follow one offboard setpoints at a time. I need to look more into this.

closed 2019-10-15 by Jaeyoung-Lim