PX4/PX4-SITL_gazebo-classic #321: Check magnetic table readout

CarlOlsson, 2019-05-24:
There was[ a bug in ECL](https://github.com/PX4/ecl/pull/606) regarding the readout of the magnetic tables sometimes resulting in a large error in magnetic declination and inclination. We need to check that the same error is not present in SITL

Jaeyoung-Lim, 2020-04-21:
@CarlOlsson Any updates regarding this issue?

CarlOlsson, 2020-04-21:
No, I think we should use the same library for calculating the magnetic field in ECL and SITL. I.e. moving the ecl geo functionality to its own submodule and also include that in SITL would probably be the best

dagar, 2020-04-21:
It's perhaps a bit convoluted, but you could include ecl in sitl_gazebo and use only the ecl_geo library.

CarlOlsson, 2020-04-21:
either that or include ecl_geo in ecl and sitl_gazebo, which one do you prefer?

TSC21, 2020-04-21:
I do prefer the approach of bringing `ecl_geo` to its own submodule and probably rename it, so it becomes agnostic for both `sitl_gazebo` and `ecl`.

LorenzMeier, 2021-01-02:
@Jaeyoung-Lim Could you have a look?