PX4/PX4-SITL_gazebo-classic #322: Roslaunch px4 posix_sitl.launch fails

tuloski, 2019-05-27:
Compiling and running SITL with "make px4_sitl_default gazebo_iris" works flawlessy,
but using the roslaunch "roslaunch px4 posix_sitl.launch vehicle:=iris" something fails. It seems a problem in mavlink instance. Qgroundcontrol doesn't see the vehicle, and also running mavros node fails.
Is that deprecated?

tuloski, 2019-05-28:
This is written on the console with the roslaunch: "ERROR [px4] Startup script returned with return value: 2". It seems like a script error, but actually both should use the same script "etc/init.d-posix/rcS"

TSC21, 2019-05-28:
https://dev.px4.io/en/simulation/ros_interface.html#launching-gazebo-with-ros-wrappers

tuloski, 2019-05-28:
@TSC21 it's exactly what I did, but there is an error and mavlink is not working.
This is the console output with "make px4_sitl_default gazebo_iris":
```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
INFO  [px4_work_queue] creating: wq:hp_default, priority: 88, stack: 16384 bytes
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-05-28/07_45_19.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
pxh> INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
```

This is with roslaunch:
```
px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
[ INFO] [1559029669.064604988]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1559029669.065224677]: waitForService: Service [/gazebo/set_physics_properties] has not 
[ INFO] [1559029669.090556957]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1559029669.090972377]: waitForService: Service [/gazebo/set_physics_properties] has not 
[INFO] [1559029669.401437, 0.000000]: Loading model XML from file
[INFO] [1559029669.401858, 0.000000]: Waiting for service /gazebo/spawn_sdf_model
[INFO] [1559029669.403188, 0.000000]: Calling service /gazebo/spawn_sdf_model
[ INFO] [1559029669.403990435, 0.024000000]: waitForService: Service 
[INFO] [1559029669.609412, 0.164000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1559029669.622675414, 0.180000000]: Physics dynamic reconfigure ready.
[ INFO] [1559029669.632236709, 0.188000000]: waitForService: Service 
[ INFO] [1559029669.666511248, 0.220000000]: Physics dynamic reconfigure ready.
ERROR [px4] Startup script returned with return value: 2
```

You can notice the error and that mavlink istance didn't start.

TSC21, 2019-05-28:
Start the daemon with xterm and gdb to find the problem source.

tuloski, 2019-05-28:
Not an expert of gdb nor xterm. Any tip?

tuloski, 2019-05-28:
Ok, did it but it doesn't help :).

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
[Thread 0x7ffff7fd7700 (LWP 8842) exited]
[New Thread 0x7fffe77fe700 (LWP 8852)]
[Thread 0x7fffe77fe700 (LWP 8852) exited]
[New Thread 0x7fffe77fe700 (LWP 8854)]
[Thread 0x7fffe77fe700 (LWP 8854) exited]
[New Thread 0x7fffe77fe700 (LWP 8856)]
[Thread 0x7fffe77fe700 (LWP 8856) exited]
[New Thread 0x7fffe77fe700 (LWP 8858)]
[Thread 0x7fffe77fe700 (LWP 8858) exited]
[New Thread 0x7fffe77fe700 (LWP 8860)]
  SYS_AUTOSTART: curr: 0 -> new: 10016
[Thread 0x7fffe77fe700 (LWP 8860) exited]

Thread 7 "lpwork" received signal SIGCONT, Continued.
[Switching to Thread 0x7ffff533f700 (LWP 8839)]
0x00007ffff6e2530d in nanosleep () at ../sysdeps/unix/syscall-template.S:84
84      ../sysdeps/unix/syscall-template.S: No such file or directory.

tuloski, 2019-05-28:
I noticed that the script (rcS) blocks here:

simulator start -s -c $simulator_tcp_port

It says on the console: "INFO  [simulator] Waiting for simulator to connect on TCP port 4560" and probably it is not able to connect on that port so it stops somehow.

TSC21, 2019-05-28:
@julianoes is this something you faced before?

julianoes, 2019-05-28:
I think I have tested this. @tuloski are all the pieces updated? You need to have the latest `sitl_gazebo` as well as PX4.

Have a look at these docs too: https://dev.px4.io/master/en/simulation/#lockstep-simulation

tuloski, 2019-05-28:
Yes everything updated to latest. And I'm reading this: https://github.com/PX4/avoidance/issues/172

It seems the (bad) solution is updating from gazebo7 to gazebo8 :( which I don't want to do.
Still don't understand why and I'm not the only person with this problem.

tuloski, 2019-05-28:
Is it normal that the tcp port for the simulator is 4560 instead of the old 14560? It seems to be missing the initial 1...sort of typo.

julianoes, 2019-05-28:
What's the world file that you are using? Have you changed it?

With lockstep we switched from UDP port 14560 to TCP port 4560, so this change is intentional.

Can you check the log output of `gzserver`. I think you can find it in `~/.gazebo`.

tuloski, 2019-05-29:
@julianoes The world is the standard one. No changes at all.
Ok for the port.
The log of gzserver are basically empty...no useful info there.
I'd like to understand the differences between launching gazebo with the "make" and with the launch, bu I have no idea how the make does everything. Any pointer?

julianoes, 2019-05-29:
>  The log of gzserver are basically empty...no useful info there.

Have you double checked all logfiles in `~/.gazebo`?

julianoes, 2019-05-29:
Make calls https://github.com/PX4/Firmware/blob/master/Tools/sitl_run.sh.

tuloski, 2019-05-29:
I found the logs.
With make:
```
(1559118250 598815312) [Msg] Waiting for master.
(1559118250 602410516) [Msg] Connected to gazebo master @ http://127.0.0.1:11345
(1559118250 602774493) [Msg] Publicized address: 10.0.0.15
(1559118250 953815644) Init world[default]
(1559118250 971476789) [Wrn] [gazebo_barometer_plugin.cpp:66] [gazebo_barometer_plugin] Using default home altitude of 488 m
(1559118250 976279839) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[0] no joint control will be performed for this channel.
(1559118250 976360149) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[1] no joint control will be performed for this channel.
(1559118250 976393496) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[2] no joint control will be performed for this channel.
(1559118250 976428958) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[3] no joint control will be performed for this channel.
(1559118250 976499888) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::propeller_joint] not found for channel[4] no joint control for this channel.
(1559118250 976726828) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_left_joint] not found for channel[5] no joint control for this channel.
(1559118250 976884595) [Wrn] [gazebo_mavlink_interface.cpp:124] joint [zephyr_delta_wing::flap_right_joint] not found for channel[6] no joint control for this channel.
(1559118250 977003363) [Dbg] [gazebo_mavlink_interface.cpp:135] <joint_name> not found for channel[7] no joint control will be performed for this channel.
(1559118250 977033442) [Msg] Connecting to PX4 SITL using TCP
(1559118250 977050181) [Msg] Lockstep is enabled
(1559118250 977066142) [Msg] Speed factor set to: 1
(1559118253 404156284) [Msg] Using MAVLink protocol v2.0
```

With launch:
```
(1559118489 470101687) [Msg] Waiting for master.
(1559118489 471025072) [Msg] Connected to gazebo master @ http://127.0.0.1:11345
(1559118489 471160758) [Msg] Publicized address: 10.0.0.15
(1559118489 793715186) Init world[default]
(1559118490 31318816) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_multirotor_base_plugin.so: libgazebo_multirotor_base_plugin.so: cannot open shared object file: No such file or directory
(1559118490 31473313) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
(1559118490 31590748) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
(1559118490 31723211) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
(1559118490 31841131) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_motor_model.so: libgazebo_motor_model.so: cannot open shared object file: No such file or directory
(1559118490 31987956) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_gps_plugin.so: libgazebo_gps_plugin.so: cannot open shared object file: No such file or directory
(1559118490 32145012) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_magnetometer_plugin.so: libgazebo_magnetometer_plugin.so: cannot open shared object file: No such file or directory
(1559118490 32294052) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_barometer_plugin.so: libgazebo_barometer_plugin.so: cannot open shared object file: No such file or directory
(1559118490 32444050) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_mavlink_interface.so: libgazebo_mavlink_interface.so: cannot open shared object file: No such file or directory
(1559118490 32601421) [Err] [Plugin.hh:165] Failed to load plugin libgazebo_imu_plugin.so: libgazebo_imu_plugin.so: cannot open shared object file: No such file or directory
```

tuloski, 2019-05-29:
Ok finally solved the mistery. Thanks everyone:

On the old instructions it was written to do "cd Firmware" and "source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default", but with the new version the folder in build is not anymore posix_sitl_default but px4_sitl_default!
So changing that source gazebo was able to find the .so.

I suggest to include in the build or in Firmware a script that does that with the correct paths.

TSC21, 2019-05-29:
> Ok finally solved the mistery. Thanks everyone:
>
> On the old instructions it was written to do "cd Firmware" and "source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default", but with the [...]

That does what in specific? If you follow the correct instructions this shouldn't happen.

tuloski, 2019-05-29:
@TSC21 
just an idea, but I'm not expert about this, but maybe the make generates a .sh that does:
source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default 
export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/sitl_gazebo/

My problem is that I had a script made by myself doing that, but then the path changed and it's easy to forget.

julianoes, 2019-05-29:
@tuloski which instructions are you referring to? Can you please paste a link.

tuloski, 2019-05-29:
@julianoes https://dev.px4.io/v1.8.0/en/simulation/ros_interface.html

closed 2019-05-29 by tuloski

julianoes, 2019-05-29:
I think this is supposed to be correct for 1.8.2. It seems like you're actually on master or 1.9.0 and therefore the instructions are not matching.

tuloski, 2019-05-29:
Yeah the documentation is correct. My problem was that I had a script from 1.8.2 doing "source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/posix_sitl_default" instead of "source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default", because in 1.9.0 and master the folder was changed from posix_sitl_default to px4_sitl_default.