PX4/PX4-SITL_gazebo-classic #324: How to disable submodule update

alexmozz, 2019-05-31:
Hi,

When I do some local changes to for instance .sdf files in models and I run `make px4_sitl_default gazebo` the changes are gone. 

What is the best way to switch off submodule update so the changes don't get overwritten?
Or maybe it is me doing something wrong, but I assume that I should run 'make' after doing modifications to .sdf files?

Thanks

TSC21, 2019-05-31:
I suppose you are doing changes on the iris.sdf file, which is the wrong way of changing the model since it's generated from xacro in every build. It has nothing to do with submodule update.

mrsiekierka, 2019-06-01:
@TSC21 Thanks for your comment. It clarifies a lot!

alexmozz, 2019-06-01:
@TSC21 Thank you for your comment. It clarifies a lot.

closed 2019-06-01 by alexmozz