PX4/PX4-SITL_gazebo-classic #327: How can i create new worlds in gazebo_sitl and mavros

sergioma295, 2019-06-08:
Hello everyone. I’m working with mavros and gazebo sitl. I’d like to know how can i create or install new worlds in order to use them after. When i create a new world in /Tools/sitl_gazebo/worlds and after i use roslaunch px4 posix_sitl.launch (i have modified the default world name to my new world) system cannot open it. Thanks so much for your attention

julianoes, 2019-06-11:
What do you mean with "system cannot open it"? What's the error message?

bozkurthan, 2019-06-12:
Actually, It is good to start with changing default worlds and adding new objects into e.g. "sitl_gazebo/worlds/iris.world". After that, In my opinion, you should understand structure. Gazebo and PX4 have some dependencies in background when you apply your command.

Can you post your console output?

TSC21, 2020-03-25:
Closed as stale.

closed 2020-03-25 by TSC21