PX4/PX4-SITL_gazebo-classic #331: Help understanding gazebo plugins in SITL

tuloski, 2019-06-20:
I'm having troubles understanding how gazebo plugins are integrated in the SITL software.
Appartently the plugin (for example the GPS plugin) retrieves all the data from gazebo "api" and it should publish some topics (/gps and /groundtruth in GPS plugin).
But actually I can't see any topic published. Only the manipulated data from MAVROS plugins.

So how is the whole chain of data working?
Is there any documentation I didn't see?

tuloski, 2019-06-20:
Oh, I see that the plugins publish topics on gazebo topics and not ros topics.
Then who is reading the gazebo topics?

TSC21, 2019-10-11:
`sitl_gazebp` has plugins that are compiled to be used within Gazebo. The topics published are used by other plugins. So plugin interact with plugins and with the simulation physics, and can exchange information using topics on the Gazebo Transport layer.

closed 2019-10-11 by TSC21