PX4/PX4-SITL_gazebo-classic #333: Help regarding dualrotor tailsitter

akshitj1, 2019-07-02:
Hi, I am trying to find simulation files for dual rotor tailsitter as described in [Design and Implementation of an Unmanned Tail-sitter](http://people.inf.ethz.ch/pomarc/pubs/BapstIROS15.pdf) by  @RomanBapst et. al. . From what I can find, in tailsitter/tailsitter.sdf there are 4 rotors.

xdwgood, 2019-09-06:
Hello @akshitj1  
Do you find dual rotor simulation files? 
I see this: https://github.com/akshitj1/sitl_gazebo  :+1: 
I try `make px4_sitl gazebo_hummingbird_fixed`....But the compilation failed, I will be grateful if you have your reply. :smiley:

akshitj1, 2019-09-06:
Check [`original repo`](https://github.com/utiasSTARS/hummingbird_ws). These guys are trying to create opensource platform for tailsitter research. The controller is very naive but will serve as good starting point for simulation files. It seems to be a big overhead if someone just wants to simulate tailsitter (for eg. [`rotors_simulator`](https://github.com/ethz-asl/rotors_simulator)

xdwgood, 2019-09-06:
@akshitj1  thank you :smiley: 
I will look at this： https://github.com/utiasSTARS/hummingbird_ws   :+1:

Jaeyoung-Lim, 2020-01-06:
@xdwgood The link doesn't exist anymore...Do you have any pointers? I would be interested in getting the control surfaces of the duo tailsitter working in this repo

xdwgood, 2020-01-06:
@Jaeyoung-Lim  I can upload my local reservation, but to be honest I did not run it successfully, hope you can get good results.

here:  https://github.com/xdwgood/Firmware/tree/jw_fix_all

Someone from my team seems to have done the job, but I heard that only five seconds can be flown, and the flight in gazebo will crash after five seconds.

akshitj1, 2020-01-06:
@Jaeyoung-Lim , I see you are from ETH zurich. There was a paper on tailsitter control by Robin Ritz from your college. links: 
[`A global controller for flying wing tailsitter vehicles`](https://www.flyingmachinearena.ethz.ch/wp-content/uploads/ritzIEEE17.pdf)
[`Additional material in Robin's thesis in Appendix 5`](https://pdfs.semanticscholar.org/0668/3817ccd1aca8437e40443fe6c3fd199931c1.pdf)

Jaeyoung-Lim, 2020-01-06:
@akshitj1 The problem is the current sitl_gazebo is not that we don't understand the dynamics, but the how to model the prop wash for the lift drag plugin. The current implementation thinks the control surfaces should stall during hover

akshitj1, 2020-01-06:
@Jaeyoung-Lim , Regarding broken link, I think repo has been updated/renamed to: [PhoenixDrone-sitl_gazebo](https://github.com/utiasSTARS/PhoenixDrone-sitl_gazebo) . I haven't seen into the code but let me know if it helps.

Jaeyoung-Lim, 2020-01-06:
@akshitj1 I looked into the code and the repo you have forwarded only models the dynamics during hover, therefore a transition will not be possible

xdwgood, 2020-02-20:
@Jaeyoung-Lim   Do you continue it? Thank you

Jaeyoung-Lim, 2020-02-20:
@xdgwood Currently not working on it

closed 2020-04-25 by LorenzMeier