PX4/PX4-SITL_gazebo-classic #334: Launching another robot with sitl_gazebo

Ahmad-Drak, 2019-07-02:
So I've been trying to launch another robot along with the typhoon 480 in gazebo.
It can be done with some help from [here](https://answers.ros.org/question/41433/multiple-robots-simulation-and-navigation/).
Main issue is typhoon 480 is launched using the make command
`make px4_sitl_default gazebo_typhoon_h480`
and I cant seem to find any launch files for it.

Can someone please explain what happens when the above command is run? 
Is there a way to make another robot (PR2 for example) launch along with the drone?

Ahmad-Drak, 2019-07-02:
If i launch the typhoon `make px4_sitl_default gazebo_typhoon_h480` in a terminal followed by the PR2 `roslaunch pr2_no_arms.launch` in another terminal, it just stays waiting for the service indefinitely

`[INFO] [1562070082.078183]: Waiting for service /gazebo/spawn_urdf_model` 

If I launch the PR2 first followed by the typhoon i get the error:

`[Err] [Master.cc:96] EXCEPTION: Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.`

hamishwillee, 2019-07-02:
I don't know the answer to your question, but the docs on multi-vehicle simulation in Gazebo for PX4 are here: https://dev.px4.io/master/en/simulation/multi-vehicle-simulation.html

mrsiekierka, 2019-07-08:
@Ahmad-Drak can you try to run your typhoon through a launch file instead of 'make'? I believe there is no need to make it every time you want to run it. You can then include pr2 launch in your typhoon launch file.

Ahmad-Drak, 2019-07-08:
@mrsiekierka I tried just that for the past week, with very weird results. I get the pr2 spawned only, but with the dynamics of the drone.

Im trying to launch the typhoon alone (instead of iris) in this launch file [here](https://github.com/PX4/Firmware/blob/master/launch/single_vehicle_spawn.launch). I edited all the 4 xacro files to account for typhoon not iris.

Well the typhoon launches, but its very very out of shape (see image below)..much larger than the world its in. If i zoom-out it just becomes dark/black.. Any suggestions how it can be fixed?

[image]

lamping7, 2019-07-15:
I'm not sure what is the issue. Are you wanting to use ROS? Do you want MAVROS also? `roslaunch px4 mavros_posix_sitl.launch vehicle:=typhoon_h480` will launch it alone.

Ahmad-Drak, 2019-07-16:
> Are you wanting to use ROS? Do you want MAVROS also?
Yes and yes
That command works. Alternatively I changed the vehicle arg in the launch file to typhoon_h480

closed 2020-04-25 by LorenzMeier

bharath5673, 2023-01-24:
> If i launch the typhoon `make px4_sitl_default gazebo_typhoon_h480` in a terminal followed by the PR2 `roslaunch pr2_no_arms.launch` in another terminal, it just stays waiting for the service indefinitely
> 
> `[INFO] [1562070082.078183]: Waiting for service /gazebo/spawn_urdf_model`
> 
> If I launch the PR2 first followed by the typhoon i get the error:
> 
> `[Err] [Master.cc:96] EXCEPTION: Unable to start server[bind: Address already in use]. There is probably another Gazebo process running.`
> 
> Same warning/errors appear when i start gazebo alone then launch both typhoon and PR2

i faced the same issue
and i solved by 

```
killall gzserver
killall gzclient
```