PX4/PX4-SITL_gazebo-classic #335: How to make IRIS more powerful.

tuloski, 2019-07-08:
Hi, I'd like to make the IRIS model more powerful, because now I reach thrust saturation when going at around 10m/s.
Is the process simple or do I need to touch the rotor_description?

RicardoM17, 2019-09-09:
@tuloski not sure if you still have this issue but you can change the parameters on the iris.sdf to "solve" this.

Either increase the thrust coefficient or the max speed.

tuloski, 2019-09-10:
Thanks. Yeah I increased both and it worked. I don't know if there is a limit or some stability issues.

RicardoM17, 2019-09-20:
There are some issues if you increase them too much. You can also decrease the weight somewhat but all of these are just adaptations and not really the ideal solution.

Still this can be closed I believe @TSC21 .

closed 2019-09-22 by TSC21