PX4/PX4-SITL_gazebo-classic #337: Gimbal plugin macro

mrsiekierka, 2019-07-10:
Hey,

I'm trying to add a gimbal controller when running multiple drones. After adding a macro I can see topics:
/gazebo/default/drone/gimbal_pitch_cmd
/gazebo/default/drone/gimbal_pitch_status
/gazebo/default/drone/gimbal_roll_cmd
/gazebo/default/drone/gimbal_roll_status
/gazebo/default/drone/gimbal_yaw_cmd
/gazebo/default/drone/gimbal_yaw_status

but there is nothing being published on those topics.

I'm assuming something is wrong with the links or possibly with the imu topic?

I'd appreciate any help! 

In component_snippets.xacro:
```
  <!-- Macro to add a gimbal controller. -->
  <xacro:macro name="gimbal_plugin_macro"
    params="namespace">
    <gazebo>
      <plugin name='gimbal_controller' filename='libgazebo_gimbal_controller_plugin.so'>
        <robotNamespace>${namespace}</robotNamespace>
        <joint_roll>camera_joint</joint_roll>
        <control_gimbal_channels>
          <channel>
            <joint_control_pid>
              <p>10.5</p>
              <i>0.0</i>
              <d>0.0</d>
              <iMax>0</iMax>
              <iMin>0</iMin>
              <cmdMax>0.3</cmdMax>
              <cmdMin>-0.3</cmdMin>
            </joint_control_pid>
            <joint_axis>joint_roll</joint_axis>
          </channel>
        </control_gimbal_channels>
        <gimbal_imu>camera_imu</gimbal_imu>
      </plugin>
    </gazebo>
  </xacro:macro>

  <!-- Macro to add a depth camera (RealSense D435). -->
  <xacro:macro name="depth_camera_macro"
    params="namespace parent_link camera_suffix frame_rate
      horizontal_fov image_width image_height image_format min_distance
      max_distance noise_mean noise_stddev enable_visual point_cloud_cut_off focal_length
      Cx Cy *cylinder *origin">
    <link name="camera_link">
      <sensor name="camera_imu" type="imu">
        <always_on>1</always_on>
      </sensor>
      <collision>
        <origin xyz="0 0 0" rpy="0 0 0" />
        <geometry>
          <xacro:insert_block name="cylinder" />
        </geometry>
      </collision>
      <xacro:if value="${enable_visual}">
        <visual>
          <origin xyz="0 0 0" rpy="0 1.57079632679 0" />
          <geometry>
            <xacro:insert_block name="cylinder" />
          </geometry>
          <material name="red" />
        </visual>
      </xacro:if>
      <inertial>
        <mass value="1e-5" />
        <origin xyz="0 0 0" rpy="0 0 0" />
        <inertia ixx="1e-6" ixy="0" ixz="0" iyy="1e-6" iyz="0" izz="1e-6" />
      </inertial>
    </link>
    <joint name="${namespace}/camera_joint" type="continuous">
      <xacro:insert_block name="origin" />
      <parent link="${parent_link}" />
      <child link="camera_link" />
      <pose>0.1 0.0 0.0 0 0 0</pose>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <limit lower="-1.57" upper="1.57" effort="0.5" velocity="3"/>
      <axis xyz="1 0 0"/>
    </joint>
    <gazebo reference="${namespace}/camera_link">
      <sensor type="depth" name="${namespace}_camera">
        <always_on>true</always_on>
        <update_rate>${frame_rate}</update_rate>
        <camera name="head">
          <horizontal_fov>${horizontal_fov}</horizontal_fov>
          <image>
            <width>${image_width}</width>
            <height>${image_height}</height>
            <format>${image_format}</format>
          </image>
          <depth_camera>
            <output>1</output>
          </depth_camera>
          <clip>
            <near>${min_distance}</near>
            <far>${max_distance}</far>
          </clip>
          <noise>
            <type>gaussian</type>
            <!-- Noise is sampled independently per pixel on each frame.
                 That pixel's noise value is added to each of its color
                 channels, which at that point lie in the range [0,1]. -->
            <mean>${noise_mean}</mean>
            <stddev>${noise_stddev}</stddev>
          </noise>
        </camera>
        <plugin name="${namespace}_camera_controller" filename="libgazebo_ros_openni_kinect.so">
          <robotNamespace>${namespace}</robotNamespace>
          <alwaysOn>true</alwaysOn>
          <baseline>0.2</baseline>
          <updateRate>${frame_rate}</updateRate>
          <cameraName>/camera</cameraName>
          <imageTopicName>camera/color/image_raw</imageTopicName>
          <cameraInfoTopicName>camera/depth/camera_info</cameraInfoTopicName>
          <depthImageTopicName>camera/depth/image_rect_raw</depthImageTopicName>
          <depthImageInfoTopicName>camera/depth/camera_info</depthImageInfoTopicName>
          <pointCloudTopicName>camera/depth/points</pointCloudTopicName>
          <frameName>camera_${camera_suffix}_link</frameName>
          <pointCloudCutoff>${point_cloud_cut_off}</pointCloudCutoff>
          <distortionK1>0.00000001</distortionK1>
          <distortionK2>0.00000001</distortionK2>
          <distortionK3>0.00000001</distortionK3>
          <distortionT1>0.00000001</distortionT1>
          <distortionT2>0.00000001</distortionT2>
          <CxPrime>0</CxPrime>
          <Cx>${Cx}</Cx>
          <Cy>${Cy}</Cy>
          <focalLength>${focal_length}</focalLength>
          <hackBaseline>0</hackBaseline>
        </plugin>
      </sensor>
    </gazebo>
  </xacro:macro>
```

_base.xacro

`  
  <xacro:gimbal_plugin_macro
    namespace="${namespace}">
  </xacro:gimbal_plugin_macro>

  
  <xacro:depth_camera_macro
    namespace="${namespace}"
    parent_link="base_link"
    camera_suffix=""
    frame_rate="15"
    horizontal_fov="1.158899"
    image_width="1280"
    image_height="720"
    image_format="R8G8B8"
    min_distance="0.02"
    max_distance="700"
    noise_mean="0.0"
    noise_stddev="0.007"
    enable_visual="true"
    point_cloud_cut_off="0.5"
    focal_length="1.93"
    Cx="640"
    Cy="360">
    <cylinder length="0.01" radius="0.007" />
    <origin xyz="0.15 0.0 0.0" rpy="0 0 0" />
  </xacro:depth_camera_macro>`

Jaeyoung-Lim, 2020-03-10:
@pawelpetruch Did you manage to solve the problem?

Jaeyoung-Lim, 2020-08-12:
Fixed in https://github.com/PX4/sitl_gazebo/pull/549

closed 2020-08-12 by Jaeyoung-Lim