PX4/PX4-SITL_gazebo-classic #338: Gazebo freezes when trigger camera of Typhoon H480 model

SwiftGust, 2019-08-13:
I'm trying to use gimbal camera of Typhoon H480 model in SITL, having problem with triggering camera.

The problem is as soon as I trigger camera from QGC UI while in ground
 QGC disconnected with vehicle and gazebo simulator also freezed, not responding.

Suspecting geotagged image plugin since last message printed different from usual SITL is
`[Dbg] [gazebo_geotagged_images_plugin.cpp:448] Handle Start Capture`

[image]

Below is console output with launching 
`make px4_sitl gazebo_typhoon_h480`
 
Thank you.
```
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.218
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
[Err] [Plugin.hh:187] Failed to load plugin libgazebo_gst_camera_plugin.so: libgazebo_gst_camera_plugin.so: cannot open shared object file: No such file or directory
[Msg] Camera on udp port 14530
[Wrn] [gazebo_sonar_plugin.cpp:75] [gazebo_sonar_plugin] Please specify a robotNamespace.
[Msg] Home latitude is set to 37.4890038.
[Msg] Home longitude is set to 126.6262883.
[Wrn] [gazebo_barometer_plugin.cpp:66] [gazebo_barometer_plugin] Using default home altitude of 488 m
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_4_joint] found for channel[0] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_5_joint] found for channel[1] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_2_joint] found for channel[2] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_3_joint] found for channel[3] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_0_joint] found for channel[4] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [rotor_1_joint] found for channel[5] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [typhoon_h480::cgo3_camera_joint] found for channel[6] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [typhoon_h480::cgo3_camera_joint] found for channel[7] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [typhoon_h480::cgo3_vertical_arm_joint] found for channel[8] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [left_leg_joint] found for channel[9] joint control active for this channel.
[Dbg] [gazebo_mavlink_interface.cpp:129] joint [right_leg_joint] found for channel[10] joint control active for this channel.
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
10.10
SITL COMMAND: "/home/seunghwan/Firmware/build/px4_sitl_default/bin/px4" "/home/seunghwan/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/seunghwan/Firmware"/test_data
INFO  [px4] Creating symlink /home/seunghwan/Firmware/ROMFS/px4fmu_common -> /home/seunghwan/Firmware/build/px4_sitl_default/tmp/rootfs/etc
0 WARNING: setRealtimeSched failed (not run as root?)

______  __   __    ___ 
| ___ \ \ \ / /   /   |
| |_/ /  \ V /   / /| |
|  __/   /   \  / /_| |
| |     / /^\ \ \___  |
\_|     \/   \/     |_/

px4 starting.

INFO  [px4] Calling startup script: /bin/sh etc/init.d-posix/rcS 0
+ SYS_AUTOSTART: curr: 10016 -> new: 6011
+ RTL_DESCEND_ALT: curr: 10.0000 -> new: 5.0000
+ RTL_LAND_DELAY: curr: 0.0000 -> new: 5.0000
* RTL_DESCEND_ALT: curr: 5.0000 -> new: 10.0000
* RTL_LAND_DELAY: curr: 5.0000 -> new: 0.0000
+ MC_PITCHRATE_P: curr: 0.2000 -> new: 0.1000
+ MC_ROLLRATE_P: curr: 0.2000 -> new: 0.0500
  MPC_XY_VEL_I: curr: 0.0200 -> new: 0.2000
+ MPC_XY_VEL_P: curr: 0.2000 -> new: 0.1500
  TRIG_INTERFACE: curr: 4 -> new: 3
  TRIG_MODE: curr: 0 -> new: 4
  MNT_MODE_IN: curr: -1 -> new: 0
  MAV_PROTO_VER: curr: 0 -> new: 2
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
INFO  [simulator] Waiting for simulator to connect on TCP port 4560
INFO  [simulator] Simulator connected on TCP port 4560.
[Msg] Using MAVLink protocol v2.0
[Msg] GimbalControllerPlugin::Init
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
Gazebo multi-robot simulator, version 9.10.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 192.168.1.218
  MAV_TYPE: curr: 2 -> new: 13
INFO  [init] Mixer: etc/mixers/hexa_x.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550
[Err] [InsertModelWidget.cc:433] Missing model.config for model "/home/seunghwan/osrf-gazebo-models/.hg"
INFO  [mavlink] mode: Onboard, data rate: 4000000 B/s on udp port 14580 remote port 14540
INFO  [mavlink] mode: Onboard, data rate: 4000 B/s on udp port 14558 remote port 14530
INFO  [logger] logger started (mode=all)
INFO  [logger] Start file log (type: full)
INFO  [logger] Opened full log file: ./log/2019-08-13/09_10_41.ulg
INFO  [mavlink] MAVLink only on localhost (set param MAV_BROADCAST = 1 to enable network)
INFO  [px4] Startup script returned successfully
[Wrn] [ModelDatabase.cc:212] Unable to connect to model database using [//database.config]. Only locally installed models will be available.
INFO  [mavlink] partner IP: 127.0.0.1
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
[Wrn] [GuiIface.cc:122] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&File" under id 56
[Wrn] [GuiIface.cc:122] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Edit" under id 59
[Wrn] [GuiIface.cc:122] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Camera" under id 61
[Wrn] [GuiIface.cc:122] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&View" under id 64
[Wrn] [GuiIface.cc:122] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Window" under id 68
[Wrn] [GuiIface.cc:122] void DBusMenuExporterPrivate::addAction(QAction*, int): Already tracking action "&Help" under id 73
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/typhoon_h480/motors, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/typhoon_h480/motor_speed/4, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/typhoon_h480/motor_speed/5, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/typhoon_h480/motor_speed/2, deleting message. This warning is printed only once.
[Wrn] [msgs.cc:1852] Conversion of sensor type[sonar] not supported.
INFO  [ecl/EKF] EKF aligned, (pressure height, IMU buf: 22, OBS buf: 14)
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/typhoon_h480/motor_speed/3, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/typhoon_h480/motor_speed/0, deleting message. This warning is printed only once.
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/typhoon_h480/motor_speed/1, deleting message. This warning is printed only once.
[Dbg] [gazebo_geotagged_images_plugin.cpp:508] Send camera info
[Dbg] [gazebo_geotagged_images_plugin.cpp:539] Send camera settings
[Dbg] [gazebo_geotagged_images_plugin.cpp:539] Send camera settings
[Dbg] [gazebo_geotagged_images_plugin.cpp:590] Send storage info
[Dbg] [gazebo_geotagged_images_plugin.cpp:569] Send capture status
INFO  [ecl/EKF] EKF GPS checks passed (WGS-84 origin set)
INFO  [ecl/EKF] EKF commencing GPS fusion
[Dbg] [gazebo_geotagged_images_plugin.cpp:448] Handle Start Capture
INFO  [mavlink] partner IP: 127.0.0.1
[Wrn] [Publisher.cc:141] Queue limit reached for topic /gazebo/default/user_camera/pose, deleting message. This warning is printed only once.

```

Jaeyoung-Lim, 2019-08-19:
@SwiftGust Any updates on debugging the issue?

SwiftGust, 2019-08-19:
With some debugging it looks like gazebo freezes when handle_take_photo function ending but no clue why yet.

Jaeyoung-Lim, 2019-10-10:
@SwiftGust Is this still an issue? I can run the camera capture without any issues

SwiftGust, 2019-10-15:
@Jaeyoung-Lim It has no issue on a new desktop
The desktop I used was maybe too slow to run this plugin.

closed 2019-10-15 by SwiftGust