PX4/PX4-SITL_gazebo-classic #347: "Already tracking action "&File" under id 56"--About px4_sitl debugging with gdb.

pxfdale, 2019-09-24:
Hi, masters of UAV! Help!
I followed the tutorial [https://dev.px4.io/master/en/debug/simulation_debugging.html#start-combinations](url).
However, when I run `make px4_sitl_default gazebo_tailsitter_gdb`, the problem are as follows:
[image]
[image]

Does anyone know the reason? Wish to get help from you.

julianoes, 2020-03-11:
I have never seen this before, no idea.
Can you just type `run` and hit enter or does it not work at all?

LorenzMeier, 2021-01-02:
Closing as stale, no reply.

closed 2021-01-02 by LorenzMeier