PX4/PX4-SITL_gazebo-classic #363: Make Error..

netkimjh, 2019-10-19:
Hi,
I installed Ubuntu 16.04, ROS Kinetic, Gazebo, and PX4.
However, there is an error as shown in the following.
Please help me to resolve this.
Thank you in advance.

```
$ sudo make px4_sitl_default gazebo

[1/5] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- Using C++14 compiler
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
--   chrono
--   date_time
--   atomic
-- Boost version: 1.58.0
-- Checking for module 'gstreamer-1.0 >= 1.0'
--   
-- Checking for module 'gstreamer-base-1.0 >= 1.0'
--   
-- Building klt_feature_tracker without catkin
-- Building OpticalFlow with OpenCV
-- catkin DISABLED
-- Gazebo version: 7.0
-- Found GStreamer: adding gst_camera_plugin
-- Found GStreamer: adding gst_video_stream_widget
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
    linked by target "LiftDragPlugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gst_camera_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_barometer_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_geotagged_images_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_magnetometer_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "sensor_msgs" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_irlock_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_video_stream_widget" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_controller_interface" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_wind_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "std_msgs" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gps_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_sonar_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "mav_msgs" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "nav_msgs" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_lidar_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_opticalflow_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_mavlink_interface" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_uuv_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_vision_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_multirotor_base_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_imu_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_gimbal_controller_plugin" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "physics_msgs" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo
    linked by target "gazebo_motor_model" in directory /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo

-- Configuring incomplete, errors occurred!
See also "/home/ubuntu-dell/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/home/ubuntu-dell/src/Firmware/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
FAILED: cd /home/ubuntu-dell/src/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_VISION_ESTIMATION_DATA=ON -GNinja /home/ubuntu-dell/src/Firmware/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/ubuntu-dell/src/Firmware/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
Makefile:193: recipe for target 'px4_sitl_default' failed
make: *** [px4_sitl_default] Error 1

```

TSC21, 2019-10-19:
`sudo apt-get install gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly`.

Also, there's no reason to use `sudo` on `make` commands.

closed 2019-10-19 by TSC21

netkimjh, 2019-10-20:
Hi @TSC21 
thank you for your reply.
However, it did not resolve this issue.... the same error happened.

TSC21, 2019-10-20:
Make sure that gstreamer1.0 is installed on your system. Then `make clean` and rebuild.

netkimjh, 2019-10-20:
I am sure that gstreamer1.0 is installed... see the result
```

$ sudo apt-get install gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly
[sudo] password for ubuntu-dell: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gstreamer1.0-plugins-base is already the newest version (1.8.3-1ubuntu0.3).
gstreamer1.0-plugins-good is already the newest version (1.8.3-1ubuntu0.4).
gstreamer1.0-plugins-bad is already the newest version (1.8.3-1ubuntu0.2).
gstreamer1.0-plugins-ugly is already the newest version (1.8.3-1ubuntu0.1).
0 upgraded, 0 newly installed, 0 to remove and 251 not upgraded.

```
and `$ make clean` and `$ sudo make px4_sitl_default gazebo` showed the same error.
Do you think gstreamer1.0 is the problem?

netkimjh, 2019-10-20:
Can you reopen this issue, @TSC21 ?

TSC21, 2019-10-20:
@netkimjh try `sudo apt-get install libgstreamer1.0-0 libgstreamer1.0-dev`

TSC21, 2019-10-20:
> Do you think gstreamer1.0 is the problem?

Yes:

```sh
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
GSTREAMER_LIBRARIES (ADVANCED)
```

netkimjh, 2019-10-20:
Okay. Thank you for your explanation and giving me a solution.
I tried `$ sudo apt-get install libgstreamer1.0-0 libgstreamer1.0-dev`, but it is still having an error.
I will try to resolve this issue reading error message clearly, and I will let you know if there is no more solution I can find.

Thank you!

TSC21, 2019-10-20:
> Okay. Thank you for your explanation and giving me a solution.
> I tried `$ sudo apt-get install libgstreamer1.0-0 libgstreamer1.0-dev`, but it is still having an error.
> I will try to resolve this [...]

If you are getting exactly the same error, it's because your sym links on your system may be wrong. Try `sudo ldconfig`, and make sure to clean the build and rebuild. If you are following the correct procedures, you shouldn't get build errors regarding gstreamer. The error you reported in https://github.com/PX4/sitl_gazebo/issues/363#issue-509522692 is because gstreamer was not found. Solution: install it and make sure that CMake and the compiler can find its libs. There's not much more to read on it ;)
Now, if you get another error, different from the first one reported, then that's a different issue and should be reported in a different thread.