PX4/PX4-SITL_gazebo-classic #367: How to receive gazebo camera image into opencv

bozkurthan, 2019-11-08:
Hello,

I'm trying to receive videostream which is published at **udp://localhost:5600** to process in opencv.
I integrated typhoon camera model to iris and received from qgroundcontrol.
[image]

[image]

But when I tried to receive video from opencv it couldn't be opened with this code. (Also I disabled QGC video receive plugin in this step.)
`video = cv2.VideoCapture("udp://127.0.0.1:5600")
`

I'm not expert on videostream but I understood the problem is decoding. 
And after that I tried to receive video from Gazebo: Image View(Topic Visulation) by using another video recorder software (OBS Studio) and stream into specific port.

Topic Visulation
[image]

OBS Studio settings (Also I used different version of Container format but it wasn't helpful)
[image]

[image]

I used this code for track:
[OpenCV-Object-Tracking](https://github.com/ehsangazar/OpenCV-Object-Tracking/blob/master/index.py)
By changing video capture line to this:
`video = cv2.VideoCapture("udp://127.0.0.1:9999?overrun_nonfatal=1&fifo_size=50000000")
`

But video is so delayed and has low fps and it is impossible to process in this situation.

I have 2 question for this:
1-) Is it possible to recieve video directly from opencv videocapture function?
2-) If it's not possible how can I increase efficiency of my method?

Thank you for helping.

bozkurthan, 2019-11-08:
I found useful link on [How to open a GStreamer pipeline from OpenCV with VideoWriter
](https://stackoverflow.com/questions/46219454/how-to-open-a-gstreamer-pipeline-from-opencv-with-videowriter) and [PX4 Dev Guide-Gazebo Simulation ](https://dev.px4.io/v1.9.0/en/simulation/gazebo.html)

I have this pipeline which is mentioned on Dev Guide:
```
gst-launch-1.0  -v udpsrc port=5600 caps='application/x-rtp, media=(string)video, clock-rate=(int)90000, encoding-name=(string)H264' \
! rtph264depay ! avdec_h264 ! videoconvert ! autovideosink fps-update-interval=1000 sync=false
```

I tried to write pipeline directly to opencv but I gave following error:

```
hanco@CRT-boi:~/Desktop/deneme_tahtasi$ python manual_tracker.py 
Could not open video
```

bozkurthan, 2019-12-11:
I added repo for who want to work with OpenCV and PX4 Gazebo Sim.
[OpenCV-Gazebo Repo](https://github.com/bozkurthan/PX4-Gazebo-Opencv)
Also I will add basic OpenCV tracker for example.

FYI @julianoes , @TSC21

closed 2019-12-11 by bozkurthan

francisulano, 2020-05-04:
Hi. Sorry for asking a question here. I'm trying to add a gimbaled camera to the iris, as you did. I've opened a topic at the PX4 Discuss forum (https://discuss.px4.io/t/sitl-add-gimbal-and-rgb-camera-to-iris-for-gazebo/16587). Would you mind answering there (or here if you prefer) on how you did it? Thank you.

bozkurthan, 2020-06-29:
@francisulano if you're still dealing I will planing to share model on the next week.

BugraBuyukarslan, 2020-07-04:
@bozkurthan Hi, im am trying to add camera to gazebo_plane and acces to this camera from gazebo to use it for object detection.
Konu hakkında bilgin varsa yardımcı olabilirsen çok sevinirim. Kolay gelsin.

Jaeyoung-Lim, 2020-07-04:
@BugraBuyukarslan The video stream is available on UDP at port 5600 for the plane_cam model

BugraBuyukarslan, 2020-07-04:
@Jaeyoung-Lim Thank you a lot, i see the stream from camera on QGC interface but when i launched my CV2 script it gives me "Error, [udp @ 0x55e6d06e07e0] bind failed: Address already in use".

When i stop using QGC and run my script again, nothing happens , after i make ctrl+c and it gives me that :
"(python:14494): GStreamer-CRITICAL **: 22:13:10.659: 
Trying to dispose element videoconvert0, but it is in PLAYING instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.
"

Do you know anything about it?
Thanks,

Jaeyoung-Lim, 2020-07-04:
@BugraBuyukarslan Ofcourse if you are receiving video stream on the QGC side, the address is already in use by QGC. You need to disable video streaming if you want to use the video stream from your application.

BugraBuyukarslan, 2020-07-04:
I did that and ran my script it does not work and gives me : 
[ WARN:0] global /home/bugra/opencv_build/opencv/modules/videoio/src/cap_gstreamer.cpp (1759) handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module typefind reported: Could not determine type of stream.
[ WARN:0] global /home/bugra/opencv_build/opencv/modules/videoio/src/cap_gstreamer.cpp (888) open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0] global /home/bugra/opencv_build/opencv/modules/videoio/src/cap_gstreamer.cpp (480) isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[1]+  Killed                  python circle_detect.py
What should i change in my script, i have so many questions sorry about that , any idea why it gives me that?
Thanks,

Jaeyoung-Lim, 2020-07-04:
@BugraBuyukarslan That is a issue with your script and not this repo.

EdoYogatama, 2020-09-08:
i need to init node the script, but it wont connect to ros master node. what can i do?

julianoes, 2020-09-09:
@EdoYogatama please don't necrobump. Create a new issue and give proper context.

ioatzi, 2022-07-21:
This repository Solve the problem of gazebo camera to opencv https://github.com/bozkurthan/PX4-Gazebo-Opencv

aradhyamathur, 2023-01-02:
Itried the above but I receive the following error `gi.repository.GLib.Error: gst_parse_error: no element "udpsrc" (1)`, can you please guide ?

julianoes, 2023-01-04:
@aradhyamathur please don't necrobump. Create a new issue and write down the steps that you took, so that someone trying to answer sees all the required pieces of information and can reproduce your issue.

aradhyamathur, 2023-01-04:
ok sure.

sergiimokin, 2024-05-21:
@bozkurthan Hello. could you tell me how to stream the video from the gazebo? I have been trying for a week to get video from the gazebo for processing in open cv, but it doesn’t work, then the plugin is not installed, then it does not work as it should, now I am thinking of writing a python script to receive video from the ros2 topic and stream it to capture on jetson nano in opencv.
I will be very grateful for the answer.

VTECEK9, 2024-10-27:
Still having this issue. Gstreamer solution works for Typhoon_h480, but cannot access camera feed outside of Gazebo for any other model.