PX4/PX4-SITL_gazebo-classic #377: cmake .. error

limeca, 2019-12-03:
Hi all,

I have some problem when I compile it.

CheckSymbolExists.c:(.text+0x1b): undefined reference to pthread_create
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_8b4b1.dir/build.make:97: recipe for target 'cmTC_8b4b1' failed
make[1]: *** [cmTC_8b4b1] Error 1
make[1]: Leaving directory '/home/gazebo/sitl_gazebo/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_8b4b1/fast' failed
make: *** [cmTC_8b4b1/fast] Error 2

Does someone have this problem?

pfhgetty, 2019-12-04:
I also have this problem.

Error log:
```Determining if the pthread_create exist failed with the following output:
Change Dir: /home/pfhgetty/workspace/Firmware/Tools/sitl_gazebo/build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_84f92/fast"
/usr/bin/make -f CMakeFiles/cmTC_84f92.dir/build.make CMakeFiles/cmTC_84f92.dir/build
make[1]: Entering directory '/home/pfhgetty/workspace/Firmware/Tools/sitl_gazebo/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_84f92.dir/CheckSymbolExists.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_84f92.dir/CheckSymbolExists.c.o   -c /home/pfhgetty/workspace/Firmware/Tools/sitl_gazebo/build/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_84f92
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_84f92.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_84f92.dir/CheckSymbolExists.c.o  -o cmTC_84f92 
CMakeFiles/cmTC_84f92.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_84f92.dir/build.make:97: recipe for target 'cmTC_84f92' failed
make[1]: *** [cmTC_84f92] Error 1
make[1]: Leaving directory '/home/pfhgetty/workspace/Firmware/Tools/sitl_gazebo/build/CMakeFiles/CMakeTmp'
Makefile:126: recipe for target 'cmTC_84f92/fast' failed
make: *** [cmTC_84f92/fast] Error 2```

Jaeyoung-Lim, 2020-01-06:
@pfhgetty @limeca I cannot reproduce this. Could you explain on how to reproduce this?

julianoes, 2020-01-15:
What system are you on? Ubuntu? If so, which version?

marcojob, 2020-07-11:
Having the same issue on Ubuntu 20.04, trying to solve it currently. Will comment here if I find something.

marcojob, 2020-07-12:
So I am still getting the error log `Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output: ... ` and `Determining if the function pthread_create exists in the pthreads failed with the following output: ...` (shortened but basically as was posted initially).
I am also getting `gzserver: symbol lookup error: Firmware/build/px4_sitl_default/build_gazebo/libgazebo_gps_plugin.so: undefined symbol: _ZN5boost6system15system_categoryEv`, which seems to be indicating some problem with either plugins or boost or just a consequence of the first error message. I have tried with both Gazebo 9 and 11, and also tried boost 1.71 and 1.67 so far no luck

julianoes, 2020-07-14:
@marcojob what computer/platform are you on?

marcojob, 2020-07-14:
I am using Ubuntu 20.04 and as outlined before I tried both with boost 1.71 and 1.67. Cmake also finds boost. What boost version are you guys using? I could try to compile and install a specific boost version to match yours

julianoes, 2020-07-14:
It should be whatever comes with https://github.com/PX4/Firmware/blob/master/Tools/setup/ubuntu.sh

SpencerHehl, 2020-09-24:
I ran into this same issue today using the ROS + Gazebo script. The resolution was to use the ubuntu.sh script that @julianoes linked in this thread, reboot, then the make succeeded for me. So looks like there might be something that script uses that the ROS + Gazebo one does not. Also probably of note, I use the common deps one as well so maybe that is where the missing piece should reside.

julianoes, 2020-09-25:
Thanks @SpencerHehl. I'm gonna close this. Please re-open or comment if the issue persists @marcojob.

closed 2020-09-25 by julianoes