PX4/PX4-SITL_gazebo-classic #378: Increase update rate in gazebo_mavlink_interface.cpp

waytry, 2019-12-09:
I am trying to implement a trajectory tracking in a quadcopter using PX4 SITL. PX4 communicate the quadcopter model in SITL through the plugin gazebo_mavlink_interface.cpp. I found that the max update rate is fixed to 250Hz in the [gazebo_mavlink_interface.cpp](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_mavlink_interface.cpp#L232). However, in my application I need a higher update rate, e.g. 1000Hz, to improve the tracking performance. How can I modify the __gazebo_mavlink_interface.cpp__ to achieve a higher update rate?

Thanks in advance.

Jaeyoung-Lim, 2020-01-06:
@waytry If you increase the communication bandwidth, this does not necessarily mean that the rest of the control loop that takes those messages will be faster. Why do you need such a high input rate? This will also be unrealistic for real hardware

TSC21, 2020-03-10:
@waytry any updates?

Jaeyoung-Lim, 2020-10-05:
Closing since there were no response. Please reopen if there is any updates

closed 2020-10-05 by Jaeyoung-Lim