PX4/PX4-SITL_gazebo-classic #379: Can I run sitl without ekf2?

heleidsn, 2019-12-16:
Hello, I am working on a project with PX4 and reinforcement learning, so I need to reset the gazebo world frequently. Is there any idea of using the gazebo ground truth position rather than the estimated position from ekf2? Thanks.

Jaeyoung-Lim, 2020-01-06:
@heleidsn You can get the groundtruth position directly if you are using ROS using the gazebo_ros package. You can find the information [here](http://gazebosim.org/tutorials?cat=connect_ros)

petertheprocess, 2020-03-10:
@heleidsn I think these will be helpful.
`def get_odom():
    try:
        handle = rospy.ServiceProxy('gazebo/get_model_state',GetModelState)
        response = handle('iris','ground_plane')
        return response
    except rospy.ServiceException, e:
        print "Service call failed: %s"%e`

closed 2020-03-10 by TSC21