PX4/PX4-SITL_gazebo-classic #381: SITL is not working with gazebo 11

shrit, 2019-12-17:
I have just updated my px4 repository to the master branch with sitl_gazebo submodule also the master branch. 
When I trying to execute the offboard example I am getting this error:
```
develop /meta/MAVSDK/example/offboard_velocity/build> ./offboard udp://:14540
[11:00:19|Info ] MAVSDK version: 0.22.0-5-gb5624676 (mavsdk_impl.cpp:25)
Wait for system to connect via heartbeat
[11:00:19|Info ] New system on: 127.0.0.1:14580 (udp_connection.cpp:263)
[11:00:19|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:390)
[11:00:19|Debug] Component Autopilot (1) added. (system_impl.cpp:403)
[11:00:20|Debug] Discovered 1 component(s) (UUID: 5283920058631409231) (system_impl.cpp:573)
System is ready
[11:00:20|Debug] MAVLink: info: ARMED by Arm/Disarm component command (system_impl.cpp:309)
Armed
[11:00:20|Warn ] command temporarily rejected (22). (mavlink_commands.cpp:170)
Takeoff failedCommand denied
exit 1

```

shrit, 2019-12-19:
After testing a different version of gazebo SITL is still not working,

shrit, 2019-12-19:
I have uploaded the only working version with my modification inside, youn can find it here:
https://github.com/shrit/sitl_gazebo/tree/good_sitl

julianoes, 2019-12-19:
That's odd. So why is it working for me?

shrit, 2019-12-20:
Actually, I have no idea what is causing the problem, I do not know if it can be a path issue or different model files, this is the only things I modify usually in order to obtain several quadrotors.
Even with one quadrotor, which is started by `make px4_sitl gazebo_iris` I had the same takeoff problem.
I do not know where to start debugging, Mavlink answers are not obvious.
Maybe we need an enum in the SDK to explain the cases why takeoff might fail
Do you agree?

julianoes, 2020-01-13:
I suggest that you try to just takeoff in PX4 without MAVSDK, so just do `pxh> commander takeoff` and debug it that way.

shrit, 2020-01-30:
I discovered my stupid error of forget updating the `GAZEBO_MODEL_PATH` in the new shell for this version. I will close this issue

closed 2020-01-30 by shrit