PX4/PX4-SITL_gazebo-classic #417: Version checking logic is wrong

leftytechie, 2020-02-26:
In various plugin files there are version checks that use incorrect logic that cause problems with gazebo installations with minor versions below 4.

```
#if GAZEBO_MAJOR_VERSION >= 7 && GAZEBO_MINOR_VERSION >= 4
    /// only gazebo 7.4 and above support Any
```

This logic is incorrect because it uses an AND, which disqualifies versions like 11.0 (which is greater than 7.4)

The correct logic should read like this:
```
#if GAZEBO_MAJOR_VERSION >7 || (GAZEBO_MAJOR_VERSION == 7 && GAZEBO_MINOR_VERSION >= 4)
    /// only gazebo 7.4 and above support Any
```
Some examples are gazebo_mavlink_interface.cpp and gazebo_gimbal_controller_plugin.cpp, but I think there are others.

It seems like this is still present in the latest code, and I haven't seen any pull requests about it, so I decided to make one.
I was a bit surprised an error this trivial managed to slide by for so long (not a criticism, I'm grateful for all the work done by folks on this repo, was just surprised).

TSC21, 2020-02-26:
I really doubt this is a problem. I for example had Gazebo 9.0 on my system and I was able to build and use the plugins correctly. I might be wrong but that sounds more like a CMake issue.

leftytechie, 2020-02-26:
I actually just had it give me a problem with gazebo 11.0 with gazebo_gimbal_controller_plugin.cpp, and fixing the logic fixed the problem, which is why I noticed it.

TSC21, 2020-02-26:
What OS are you running this on? What's the CMake version?

leftytechie, 2020-02-26:
I'm on Ubuntu 18.04, cmake version 3.10.2.

I understand why you think it wouldn't be a problem, because the code in the header files should match the code in the c files. It was a problem for me because I made modifications inside these logic blocks using gazebo 9.10, and then when I switched to 11.0, the code got skipped and it caused problems.

I suppose it could be possible for everyone to program around these blocks. It also seems fairly easy to fix though.

leftytechie, 2020-02-26:
Easy enough that even I could do it ;)

closed 2020-03-08 by leftytechie