PX4/PX4-SITL_gazebo-classic #419: Mavlink plugin not working with Gazebo 11.0

leftytechie, 2020-02-26:
I could be doing something wrong, but I've able to run pretty much simulation I want as long I don't use the mavlink plugin.

I am using Ubuntu 18.04, cmake 3.10.2, gazebo 11.0, boost 1.65.1.
```
[Err] [Model.cc:1112] Exception occured in the Load function of plugin with name[mavlink_interface] and filename[libgazebo_mavlink_interface.so]. This plugin will not run.
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
/home/amed/Programs/PX4/test/Firmware/Tools/sitl_run.sh: line 97: 29847 Aborted                 (core dumped) gzserver --verbose "${src_path}/Tools/sitl_gazebo/worlds/${model}.world"
```

I am using the master branch, it is up to date, and I have confirmed that it is working with gazebo 9.12.

TSC21, 2020-02-26:
@leftytechie there is nowhere written in our documentation that we support Gazebo 11.0. Specially since that would involve some substancial dependency checking and source code changes related to Ignition dependencies.

leftytechie, 2020-02-26:
I apologize, this thread confused me:
https://github.com/PX4/sitl_gazebo/issues/381

leftytechie, 2020-02-26:
For what it's worth, almost everything else appeared to be working.

closed 2020-02-26 by leftytechie

petergerten, 2020-05-26:
I got the same error with the Gazebo 11 docker container (https://github.com/PX4/containers/blob/master/docker/Dockerfile_simulation-focal)

```
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
/home/user/Firmware/Tools/sitl_run.sh: line 130:  6031 Aborted                 (core dumped) gzserver "${src_path}/Tools/sitl_gazebo/worlds/empty.world"
```

iftahnaf, 2020-12-18:
I'm running Ubuntu 20.04 with v1.11 and gazebo11 and facing the same error.
Any solution to this problem?

Jaeyoung-Lim, 2020-12-20:
```
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
```
This happens when gazebo is unable to find one of the resources. @iftahnaf Does this happen with the default models?

czp0204, 2023-05-13:
> Hello. Did you solve this problem ?