PX4/PX4-SITL_gazebo-classic #422: gimbal controller plugin yaw setpoint race condition

jannsta1, 2020-03-06:
I noticed that the yaw axis is quite jittery in the typhoon_h480 model when the aircraft rotates so that it is not aligned with its starting orientation. After some investigation I found that this is because it is possible for a new OnYawStringMsg to set `this->yawCommand` back to (typically) 0 before yawLimited is set for the route. I.e. Before:

https://github.com/PX4/sitl_gazebo/blob/8794a7067ab307d6919a85f7743abbc7836e64fd/src/gazebo_gimbal_controller_plugin.cpp#L487-L489

or 
https://github.com/PX4/sitl_gazebo/blob/8794a7067ab307d6919a85f7743abbc7836e64fd/src/gazebo_gimbal_controller_plugin.cpp#L497-L499

By replacing https://github.com/PX4/sitl_gazebo/blob/8794a7067ab307d6919a85f7743abbc7836e64fd/src/gazebo_gimbal_controller_plugin.cpp#L477 with `double this_yaw = this->yawCommand + this->lastImuYaw;` and using this temporary variable throughout the function the problem is resolved, albeit in a hacky fashion.

LorenzMeier, 2020-03-08:
Thanks for reporting this!

m1baldwin, 2020-04-02:
@jannsta1 This was helpful to fix a problem I was having with shaking in the yaw axis, thanks. Here a couple screen captures of the difference I noticed before/after making the recommended edits.

[image]
[image]

LorenzMeier, 2020-04-02:
@MaEtUgR Could you please follow up on this with @Jaeyoung-Lim ?

PR #452 (2020-04-03): Fix race condition in gimbal_controller_plugin
This Fixes #422 where the race condition on the yaw command in the `gimbal_controller_plugin` was introducing oscillations. The race conditon was caused when a new roll / pitch / yaw command was received while the gimbal control was calculating the gimbal commands in the `OnUpdate` method. 

@jannsta1 Thanks for reporting the issue!

**Additional Context**
H520 still shows some oscillations, but this seems to be introduced when the landing gear is up. Will be addressed on a separate PR
closed 2020-04-03 by LorenzMeier