PX4/PX4-SITL_gazebo-classic #424: gstreamer1.0-* install one-liner is broken

iwishiwasaneagle, 2020-03-09:
When running

```sudo apt-get install $(apt-cache --names-only search ^gstreamer1.0-* | awk '{print $1}' | grep -v gstreamer1.0-hybris) -y```

I receive the error

```
Reading package lists... Done                                                                                                                                                                             
Building dependency tree                                                                                                                                                                         
Reading state information... Done                                                                                                                                                          
gstreamer1.0-clutter-3.0 is already the newest version (3.0.26-1).                                                                                                                                      
gstreamer1.0-clutter-3.0 set to manually installed.                                                                                   
gstreamer1.0-alsa is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                         
gstreamer1.0-alsa set to manually installed.                                                                                               
gstreamer1.0-gl is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                           
gstreamer1.0-gl set to manually installed.                                                                                                                                                              
gstreamer1.0-gtk3 is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                          
gstreamer1.0-gtk3 set to manually installed.                                                                                                                                    
gstreamer1.0-packagekit is already the newest version (1.1.9-1ubuntu2.18.04.5).                                                                                                           
gstreamer1.0-packagekit set to manually installed.                                                                                                                                          
gstreamer1.0-plugins-base is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                      
gstreamer1.0-plugins-base set to manually installed.                                                                                                                                                      
gstreamer1.0-plugins-base-apps is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                 
gstreamer1.0-plugins-base-apps set to manually installed.                                                                                                                                                 
gstreamer1.0-plugins-good is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                      
gstreamer1.0-plugins-good set to manually installed.                                                                                                          
gstreamer1.0-pulseaudio is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                   
gstreamer1.0-pulseaudio set to manually installed.                                                                                            
gstreamer1.0-tools is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                       
gstreamer1.0-tools set to manually installed.                                                                                              
gstreamer1.0-x is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                        
gstreamer1.0-x set to manually installed.                                                                                                    
gstreamer1.0-libav is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                         
gstreamer1.0-plugins-bad is already the newest version (1.14.5-0ubuntu1~18.04.1).                                                                                                                       
Some packages could not be installed. This may mean that you have                                                                                                                                         
requested an impossible situation or if you are using the unstable                                                                                                                               
distribution that some required packages have not yet been created                                                                                                                                       
or been moved out of Incoming.                                                                                                                                                                          
The following information may help to resolve the situation:                                                                                                                                            
                                                                                                                                                                                                          
The following packages have unmet dependencies.                                                                                                                                                  
 gstreamer1.0-python3-dbg-plugin-loader : Conflicts: gstreamer1.0-python3-plugin-loader but 1.14.5-0ubuntu1~18.04.1 is to be installed                                                     
 gstreamer1.0-python3-plugin-loader : Conflicts: gstreamer1.0-python3-dbg-plugin-loader but 1.14.5-0ubuntu1~18.04.1 is to be installed                                                                  
E: Unable to correct problems, you have held broken packages.                                                                                                                          
```

Removing gstreamer1.0-python3-dbg-plugin-loader from the install list works fine

```
 sudo apt-get install $(apt-cache --names-only search ^gstreamer1.0-* | awk '{print $1}' | grep -v gstreamer1.0-hybris | grep -v gstreamer1.0-python3-dbg-plugin-loader)
```

However, I believe this method may be outdated as the gstreamer website doesn't say to install ALL packages but only a select 12 out of 41 [link](https://gstreamer.freedesktop.org/documentation/installing/on-linux.html?gi-language=c)

TSC21, 2020-03-10:
@iwishiwasaneagle thanks for reporting. Do you mind issuing a PR to fix the README file? The required packages should be:
- libgstreamer-plugins-base1.0-dev
- gstreamer1.0-plugins-bad 
- gstreamer1.0-plugins-base
- gstreamer1.0-plugins-good
- gstreamer1.0-plugins-ugly

iwishiwasaneagle, 2020-03-10:
@TSC21 will do

closed 2020-03-10 by iwishiwasaneagle