PX4/PX4-SITL_gazebo-classic #430: readme installation instructions out of date

andy7809, 2020-03-11:
Running 
`sudo apt-get install gazebo9 libgazebo9-dev`
fails because of unmet dependencies. See below:
`
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gazebo9 : Depends: libgazebo9 (= 9.12.0-1~bionic) but it is not going to be installed
           Depends: gazebo9-common (= 9.12.0-1~bionic) but it is not going to be installed
           Recommends: gazebo9-plugin-base
 libgazebo9-dev : Depends: libgazebo9 (= 9.12.0-1~bionic) but it is not going to be installed
                  Depends: gazebo9-common (= 9.12.0-1~bionic) but it is not going to be installed
                  Depends: gazebo9-plugin-base (= 9.12.0-1~bionic)
E: Unable to correct problems, you have held broken packages.
`
I install libgazebo9, then the packages gazebo9 and libgazebo9 are installed fine.

May I update the readme with an updated apt one liner for installation?

andy7809, 2020-03-11:
The installation also fails on make with:
```
[ 42%] Generating ../models/3DR_gps_mag/3DR_gps_mag-gen.sdf
Traceback (most recent call last):
  File "/home/andrew/code/GazeboSITL/sitl_gazebo/scripts/jinja_gen.py", line 4, in <module>
    import jinja2
ModuleNotFoundError: No module named 'jinja2'
```
I use `sudo pip3 install jinja2`and the make succeeds. This python dependency must also be missed somewhere in the build?

Jaeyoung-Lim, 2020-04-27:
@andy7809 Sorry for the late reply, would be awesome if you can update the readme thanks

closed 2021-01-10 by Jaeyoung-Lim