PX4/PX4-SITL_gazebo-classic #431: Standard VTOL loses global position in HITL

ColonelChicken5648, 2020-03-13:
I am trying to run gazebo using standard_vtol.world and qgroundcontrol with the HIL standard vtol airframe selected. The aircraft wobbles on takeoff and within 7-30 seconds it gives the errors "Critical navigation failure! Check sensor calibrations" and "Failed enable no global position."  I have run iris.world with the copter simulation airframe with no problems. While running iris.world with the standard vtol airframe the aircraft does a vertical takeoff, but gets confused when it tries to transition to fixed wing and move forward, so it lands. This leads me to believe there is a problem with the standard_vtol world and/or model files. Does anyone have any insight into this issue?

Jaeyoung-Lim, 2020-04-29:
@ColonelChicken5648 How is the flight controller connected to the simulation?

ColonelChicken5648, 2020-04-29:
@Jaeyoung-Lim The pixhawk is connected to the PC via USB.

Jaeyoung-Lim, 2020-04-29:
@ColonelChicken5648 I think what you are running is a bit outdates. standard_vtol.world had higher update rates than other worlds. But this does not exist anymore. I am wondering if this caused the issues you are seeing.

Could you maybe update to the latest master and try again?

jasonotu, 2020-06-15:
I can confirm using standard_vtol.world does not fix the issue.

Jaeyoung-Lim, 2020-06-15:
@jasonotu What baudrate are you using?

jasonotu, 2020-06-15:
@Jaeyoung-Lim the standard_vtol.sdf is using 921600

jasonotu, 2020-07-29:
Any update on this? I'm using a Pixhawk 4.