PX4/PX4-SITL_gazebo-classic #437: Plugin Usage Tutorials

McKayCyborg, 2020-03-18:
Hi there, 

We are looking at building our own custom rover with a skid steer drive system. In order to do this, I need to better understand how sensor and actuator information is communicated between gazebo and PX4 and how to leverage these plugins to implement custom sensors and actuators. I have read through the PX4 documentation on dronecode and understand it at a very high level, but there is no information on how to build a model.sdf file that utilises these packages for custom builds and the interface source code doesn't really have any commenting. 

Is there currently any documentation on how to do this? If not, when will this kind of documentation be available? It seems like creating custom simulated robots that leverage the PX4 flight stack would be a reasonably common requirement. Any help would be greatly appreciated.

hamishwillee, 2020-03-19:
@McKayCyborg No documentation as far as I know; and it will be written as soon as someone with expertise puts the time in to help with that.

@Jaeyoung-Lim Just did some work implementing the gazebo parachute plugin, and might be able to advise. Jae, do you think we could create some high level guidance so that next time this comes up I can point people at that rather than to you?

McKayCyborg, 2020-03-19:
To be clear, all I'm looking to do is create a skid steer rover, add some sensor plugins (2 x gps, imu, magnetometer, LiDAR etc) and add the mavlink interface so that I can map the yaw and thrust information to the wheel joints for some basic driving simulations. Down the track I will use the micrortps_client to interface with some ROS libraries but for the moment we just want the basic simulation for our custom robot.

Jaeyoung-Lim, 2020-03-19:
@hamishwillee If it can be useful sure!

@McKayCyborg The answer for your question for now:
- This repo contains models and plugins for gazebo. Implementing gazebo plugins are fairly well documented in the [gazebo tutorials](http://gazebosim.org/tutorials/?tut=plugins_hello_world) You can also look at other plugins in this repo for reference.
- For creating sdf models, you can refer to this [tutorial](http://gazebosim.org/tutorials?tut=build_robot&cat=build_robot) or sdf is not a gazebo specific thing, look at the [sdf specification](http://sdformat.org/spec) Attaching sensors in a sdf file is pretty straight forward.
- The skid steer rover is currently not supported in this repo(yet) There is a open [PR](https://github.com/PX4/sitl_gazebo/pull/394) here and the main blocker is that gazebo contact dynamics are unable to handle skid steering dynamics natively. In the meantime you can use the normal rover model since most of the controls / software are the same. The only difference is that steering commands don't result in rotation when the velocity is zero.

Hope this helps

McKayCyborg, 2020-03-19:
@Jaeyoung-Lim thanks very much for your reply. It's unfortunate this hasn't been implemented yet as our application is very skid-steer-specific and can't be developed with an ackerman steering vehicle. It may be hard to tell, but do you have any idea when the skid steering dynamics will be adequately modeled?

TSC21, 2020-03-19:
> @Jaeyoung-Lim thanks very much for your reply. It's unfortunate this hasn't been implemented yet as our application is very skid-steer-specific and can't be developed with an ackerman steering vehicle. The hardest part is that I need to customise the PX4 codebase to implement application-specific functionality and its impossible to have to take the vehicle out to a testing area every time I change the code.
> 
> It may be hard to tell, but do you have any idea when the skid steering dynamics will be adequately modeled?

My personal recommendation is that you try to explore that yourselves and contribute the the community. The core contributors and maintainers also have a lot on their backs at this point so we need to focus and establish priorities. Any contribution is welcomed!

McKayCyborg, 2020-03-19:
@TSC21 Yeah I totally understand. We don't have a lot of in-house experience using Gazebo so it may be over our heads, so in the short term we are going to stick with using a simple differential drive and maybe down the track when we feel like we need a more honest simulation, we will attempt to develop skid-steer ourselves and keep you guys updated on our progress.

Thanks for all your help guys. @hamishwillee @Jaeyoung-Lim @TSC21

hamishwillee, 2020-03-19:
@McKayCyborg Thank you. I am sure that @Jaeyoung-Lim and other members of the community will help you with specific questions. Closing this now.

@Jaeyoung-Lim Your first two points are basically "this is a gazebo thing". If that is the fact probably it makes sense to make that clear in our gazebo docs: http://dev.px4.io/master/en/simulation/gazebo.html (essentially your first two points above).

In theory you might also cover:
- PX4-specific integration point
- Explain limits of what gazebo can do in PX4 integration (e.g. like your bullet 3 above)

Basically, whatever you think will mean that people don't have to directly ask this question again :-)

closed 2020-03-19 by hamishwillee

McKayCyborg, 2020-03-20:
Sorry to keep posting, but where should I be asking questions about gazbo/mavlink/px4 communications? Is that something I should be posting here? @Jaeyoung-Lim 

Cheers

hamishwillee, 2020-03-20:
@McKayCyborg Probably in https://discuss.px4.io/c/px4/25 or slack in the #simulation channel.

McKayCyborg, 2020-03-20:
Thanks mate.