PX4/PX4-SITL_gazebo-classic #444: McMillan airfields model has terrain above ground

Jaeyoung-Lim, 2020-03-26:
When running the Mcmillan airfields world the terrain is above the ground. The world was added in #406 

Therefore the vehicle gets spawned under the terrain.

[image]

I tried defining the `pose` of the model, but with the heightmap it doesn't seem to move

@ahcorde Any ideas how to solve this?

tfoote, 2020-03-27:
As my comment here: https://github.com/PX4/sitl_gazebo/pull/406#issuecomment-604750827

Should the origin have altitude 0 if it's supposed to be at MSL?

TSC21, 2020-03-27:
> As my comment here: [#406 (comment)](https://github.com/PX4/sitl_gazebo/pull/406#issuecomment-604750827)
> 
> Should the origin have altitude 0 if it's supposed to be at MSL?

IMHO I think that it makes sense the elevation is set at 280m, if 280m is the AMSL. It means that the aircraft will start at 280m AMSL if it spawns at 0,0,0.

Jaeyoung-Lim, 2020-10-05:
This was fixed in the model database

closed 2020-10-05 by Jaeyoung-Lim