PX4/PX4-SITL_gazebo-classic #446: HITL with Plane

davidosten, 2020-03-26:
Hello,

I have troubles checking out the latest repository. My folder "sitl_gazebo" remains empty. Were the two projects Firmware and sitl_gazebo separated? My old make targets for plane HITL as described in https://dev.px4.io/master/en/simulation/ do actually not work, however it can also be of course a git mistake from my side. 

I am specifically using "make px4_sitl gazebo_plane" which does not work for me, but did formerly.

Any help would be highly appreciated.

Kind regards

David

hamishwillee, 2020-03-26:
@davidosten This looks like an environment setup issue - what PX4 version are you using on what OS? 

I know that the setup in http://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html works on a clean Ubuntu 18.04 build for PX4 v1.10 and later.
Looking at the error most likely you have a Python setup issue.

Now, once that is fixed, you'll discover that the plan simulation starts, but does not work properly due to mixer configuration issues.
@julianoes may be able to advise.

davidosten, 2020-03-29:
Thanks for your reply @hamishwillee !

I still face problems in building a fresh, clean git-clone of px4 gazebo plane. Setup is an Ubuntu 18.04.4 LTS, with PX4 via git clone https://github.com/PX4/Firmware.git --recursive, so I assume the latest version.

The trailing lines of my build-log are:

Traceback (most recent call last):
  File "tools/px_generate_uorb_topic_files.py", line 46, in <module>
    from packaging import version
ModuleNotFoundError: No module named 'packaging'
[17/738] Generating git version header
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

I can successfully build an earlier version which I forked as https://github.com/davidosten/Firmware.git
Building this code, no radio-control signals are received during HITL Simulation.

What can I do to fix my "No module named 'packagin' error?

hamishwillee, 2020-03-29:
The PX4 setup is fine; I **think** the problem is your Ubuntu. Essentially it is trying to pick up the packaging module from Python using Python 3 (say) and you only have that in Python 2 (say). A few people have had similar problems as we try to move fully to Python 3. This doesn't happen on a clean Ubuntu 18.04 machine.

I'm not sure what to recommend. Probably try to install the module - e.g. try `python -m pip install packaging` or `pip3 install packaging`. If you don't use Python for anything else try removing PYthon2.

julianoes, 2020-03-30:
Please paste the output of:
```
which python
which python3
which pip
which pip3
python --version
python3 --version
pip --version
pip3 --version
```

julianoes, 2020-03-30:
> sudo make px4_sitl gazebo_plane

Please don't use `sudo`, it's not required. You'll have to do `sudo make clean` before you can stop using `sudo`.

davidosten, 2020-03-30:
Thanks for the reply, here's the output. As a miniconda environment is used, the output of "conda list" is also printed below:

(np) david@Pluto:~/miniconda3/envs/np/Firmware$ conda list
packages in environment at /home/david/miniconda3/envs/np:

Name                    Version                   Build  Channel
_libgcc_mutex             0.1                        main  
ca-certificates           2019.10.16                    0  
certifi                   2019.9.11                py38_0  
cloudpickle               1.2.2                    pypi_0    pypi
cycler                    0.10.0                   pypi_0    pypi
expat                     2.2.6                he6710b0_0  
future                    0.18.2                   pypi_0    pypi
git                       2.23.0          pl526hacde149_0  
gym                       0.15.4                   pypi_0    pypi
gym-jsbsim                0.1                      pypi_0    pypi
jsbsim                    1.1.0.dev1               pypi_0    pypi
kiwisolver                1.1.0                    pypi_0    pypi
krb5                      1.16.1               h173b8e3_7  
libcurl                   7.67.0               h20c2e04_0  
libedit                   3.1.20181209         hc058e9b_0  
libffi                    3.2.1                hd88cf55_4  
libgcc-ng                 9.1.0                hdf63c60_0  
libssh2                   1.8.2                h1ba5d50_0  
libstdcxx-ng              9.1.0                hdf63c60_0  
matplotlib                3.1.2                    pypi_0    pypi
ncurses                   6.1                  he6710b0_1  
numpy                     1.17.4                   pypi_0    pypi
opencv-python             4.1.2.30                 pypi_0    pypi
openssl                   1.1.1d               h7b6447c_3  
pcre                      8.43                 he6710b0_0  
perl                      5.26.2               h14c3975_0  
pip                       19.3.1                   py38_0  
pyglet                    1.3.2                    pypi_0    pypi
pyparsing                 2.4.5                    pypi_0    pypi
python                    3.8.0                h0371630_2  
python-dateutil           2.8.1                    pypi_0    pypi
readline                  7.0                  h7b6447c_5  
scipy                     1.3.3                    pypi_0    pypi
setuptools                42.0.1                   py38_0  
six                       1.13.0                   pypi_0    pypi
sqlite                    3.30.1               h7b6447c_0  
tk                        8.6.8                hbc83047_0  
wheel                     0.33.6                   py38_0  
xz                        5.2.4                h14c3975_4  
zlib                      1.2.11               h7b6447c_3  
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ which python
/home/david/miniconda3/envs/np/bin/python
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ which python3
/home/david/miniconda3/envs/np/bin/python3
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ which pip
/home/david/miniconda3/envs/np/bin/pip
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ which pip3
/usr/bin/pip3
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ python --version
Python 3.8.0
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ python3 --version
Python 3.8.0
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ pip --version
pip 19.3.1 from /home/david/miniconda3/envs/np/lib/python3.8/site-packages/pip (python 3.8)
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$ pip3 --version
pip 9.0.1 from /usr/lib/python3/dist-packages (python 3.6)
(np) david@Pluto: ~/miniconda3/envs/np/Firmware$

julianoes, 2020-03-31:
I don't know conda but in this case make sure the pip package `packaging` is installed, however that is done in the conda world.

closed 2020-12-30 by Jaeyoung-Lim