PX4/PX4-SITL_gazebo-classic #450: SITL custom rovert.sdf causes gazebo connection issue

McKayCyborg, 2020-03-31:
Hi there, 

I am trying to make a custom rover implementation in gazebo that has a skid-steer arrangement and have copied my custom model into sitl_gazebo/models/rover/rover.sdf. After running 'make px4_sitl gazebo_rover', it appears as though the connection form the px4 daemon to Gazebo fails because a boost pointer gets called without being initialised. I'm assuming that it is only initialised when the connection is successful, so the question remains: why might this be happening?

Here is the error output:

`[0/4] Performing build step for 'sitl_gazebo'
ninja: no work to do.
[3/4] cd /home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default/tmp && /home...irmware_Test/Firmware /home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default
SITL ARGS
sitl_bin: /home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: rover
src_path: /home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware
build_path: /home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default
GAZEBO_PLUGIN_PATH :/home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /home/cde01/catkin_ws/devel/lib:/opt/ros/melodic/lib:/home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default/build_gazebo
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
/home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/Tools/sitl_run.sh: line 97: 16125 Aborted                 (core dumped) gzserver "${src_path}/Tools/sitl_gazebo/worlds/${model}.world"
SITL COMMAND: "/home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default/bin/px4" "/home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware"/test_data
INFO  [px4] Creating symlink /home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/ROMFS/px4fmu_common -> /home/cde01/Documents/PX4/Gazebo_Firmware_Test/Firmware/build/px4_sitl_default/tmp/rootfs/etc`

The rover.sdf file:

`<?xml version='1.0'?>
<sdf version='1.5'>
  <model name="rover">
  <static>false</static>

	<link name='chassis'>
    <inertial>
      <mass>15</mass>
      <inertia>
        <ixx>3</ixx>
        <ixy>0.0</ixy>
        <iyy>7</iyy>
        <ixz>0.0</ixz>
        <iyz>0.0</iyz>
        <izz>7</izz>
      </inertia>
      <pose>0 0 0 0 0 0</pose>
    </inertial>
		<pose>0 0 .1 0 0 0</pose>
		<collision name='chassis_collision'>
			<geometry>
				<box>
					<size>.5 .3 .1</size>
				</box>
			</geometry>
		</collision>

    <visual name='chassis_visual'>
			<geometry>
				<box>
					<size>.5 .3 .1</size>
				</box>
			</geometry>
		</visual>

    <collision name='front_caster_collision'>
      <pose>0.2 0 -.05 0 0 0</pose>
      <geometry>
          <sphere>
              <radius>.05</radius>
          </sphere>
      </geometry>
      <surface>
          <friction>
[... 248 lines cut ...]
          </joint_control_pid>
        </channel>
        <channel name="rear_right_wheel_drive">
          <input_index>6</input_index>
          <input_offset>0</input_offset>
          <input_scaling>40</input_scaling>
          <zero_position_disarmed>0</zero_position_disarmed>
          <zero_position_armed>0</zero_position_armed>
          <joint_control_type>velocity</joint_control_type>
          <joint_name>right_wheel_joint</joint_name>
          <joint_control_pid>
            <p>50</p>
            <i>20</i>
            <d>2</d>
            <iMax>800</iMax>
            <iMin>-800</iMin>
            <cmdMax>1200</cmdMax>
            <cmdMin>-1200</cmdMin>
          </joint_control_pid>
        </channel>
      </control_channels>
    </plugin>

  </model>
</sdf>`

Jaeyoung-Lim, 2020-04-02:
@McKayCyborg What exactly is different compared to the default `rover.sdf`?

Jaeyoung-Lim, 2020-08-12:
Closing as stale

closed 2020-08-12 by Jaeyoung-Lim