PX4/PX4-SITL_gazebo-classic #456: Run make px4_sitl gazebo_tailsitter Error

xdwgood, 2020-04-13:
gazebo version: Gazebo multi-robot simulator, version 9.11.0

`/home/dw/Firmware/Tools/sitl_gazebo/src/gazebo_usv_dynamics_plugin.cpp: In member function ‘virtual void gazebo::UsvDynamicsPlugin::Update()’:
/home/dw/Firmware/Tools/sitl_gazebo/src/gazebo_usv_dynamics_plugin.cpp:251:29: error: missing template arguments before ‘xformV’
     ignition::math::Matrix4 xformV(vq);
                             ^
/home/dw/Firmware/Tools/sitl_gazebo/src/gazebo_usv_dynamics_plugin.cpp:285:15: error: ‘xformV’ was not declared in this scope
       bpntW = xformV * bpnt;
               ^
At global scope:
cc1plus: warning: unrecognized command line option ‘-Wno-address-of-packed-member’
[73/100] Building CXX object CMakeFile...c/gazebo_geotagged_images_plugin.cpp.o
ninja: build stopped: subcommand failed.
[737/740] Linking CXX executable bin/px4
FAILED: cd /home/dw/Firmware/build/px4_sitl_default/build_gazebo && /usr/bin/cmake --build .
ninja: build stopped: subcommand failed.
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1
`

@Jaeyoung-Lim  could you have a look?

xdwgood, 2020-04-13:
I tried v1.10 and everything works fine in gazebo.

@Jaeyoung-Lim  Could you see if this problem is caused by gazebo? I will also try to do some research today.  https://github.com/PX4/Firmware/issues/14588

Jaeyoung-Lim, 2020-04-21:
@xdwgood I think this is the same issue as #464

Pedro-Roque, 2020-04-21:
@Jaeyoung-Lim yes, same issue .

xdwgood, 2020-04-29:
Any update? thanks

Jaeyoung-Lim, 2020-04-29:
@xdwgood Which system are you running? This is coming from your system, since the code is built for every PR and checked through the CI. Can you check all your ignition dependency versions?

xdwgood, 2020-04-29:
ubuntn 16.04

Jaeyoung-Lim, 2020-04-29:
@xdwgood The problem is that you are using the wrong ignition libraries, e.g. `libignition-math` please check if the versions are correct

Jaeyoung-Lim, 2020-05-05:
@xdwgood Did you manage to fix it?

xdwgood, 2020-05-05:
@Jaeyoung-Lim  Sorry， I haven‘t fix it...

Jaeyoung-Lim, 2020-05-05:
@xdwgood What is your `libignition-math` version?

Pedro-Roque, 2020-05-06:
@Jaeyoung-Lim this might be related to a minor version of math, and caused by updating the libraries (other than at the moment of installing ros-*-desktop-full), for instance when updating gazebo to latest minor version. I tried to check how to get the minor version of math, but didnt get it done yet. Could you check on it?

Jaeyoung-Lim, 2020-05-06:
@Pedro-Roque 
```
$ apt-cache policy libignition-math4
libignition-math4:
  Installed: 4.0.0+dfsg1-4
  Candidate: 4.0.0+dfsg1-4
  Version table:
 *** 4.0.0+dfsg1-4 500
        500 http://ch.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status
     4.0.0-1~bionic 500
        500 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main amd64 Packages

```

Pedro-Roque, 2020-05-06:
@Jaeyoung-Lim  This is what I have:
```
➜  ~ sudo find /usr | grep ignition-math
[sudo] password for roque: 
/usr/share/doc/libignition-math4-dev
/usr/share/doc/libignition-math4-dev/changelog.Debian.gz
/usr/share/doc/libignition-math4-dev/copyright
/usr/share/doc/libignition-math4
/usr/share/doc/libignition-math4/changelog.Debian.gz
/usr/share/doc/libignition-math4/copyright
/usr/share/doc/libignition-math4/README.md
/usr/share/doc/libignition-math4/AUTHORS
/usr/share/doc/libignition-math4/NEWS.gz
/usr/share/lintian/overrides/libignition-math4
/usr/local/share/ignition/ignition-math4_0
/usr/local/share/ignition/ignition-math4_0/ignition-math4.tag.xml
/usr/local/lib/cmake/ignition-math4
/usr/local/lib/cmake/ignition-math4/ignition-math4-targets-relwithdebinfo.cmake
/usr/local/lib/cmake/ignition-math4/ignition-math4-config.cmake
/usr/local/lib/cmake/ignition-math4/ignition-math4-targets.cmake
/usr/local/lib/cmake/ignition-math4/ignition-math4-config-version.cmake
/usr/local/lib/pkgconfig/ignition-math4.pc
/usr/local/lib/libignition-math4.so.4
/usr/local/lib/libignition-math4.so
/usr/local/lib/libignition-math4.so.4.0.0
/usr/lib/x86_64-linux-gnu/cmake/ignition-math4
/usr/lib/x86_64-linux-gnu/cmake/ignition-math4/ignition-math4-targets-relwithdebinfo.cmake
/usr/lib/x86_64-linux-gnu/cmake/ignition-math4/ignition-math4-config.cmake
/usr/lib/x86_64-linux-gnu/cmake/ignition-math4/ignition-math4-targets.cmake
/usr/lib/x86_64-linux-gnu/cmake/ignition-math4/ignition-math4-config-version.cmake
/usr/lib/x86_64-linux-gnu/pkgconfig/ignition-math4.pc
/usr/lib/x86_64-linux-gnu/libignition-math4.so.4
/usr/lib/x86_64-linux-gnu/libignition-math4.so
/usr/lib/x86_64-linux-gnu/libignition-math4.so.4.0.0

```

and 

```
➜  ~ apt-cache policy libignition-math4        
libignition-math4:
  Installed: 4.0.0+dfsg1-4
  Candidate: 4.0.0+dfsg1-4
  Version table:
 *** 4.0.0+dfsg1-4 500
        500 http://se.archive.ubuntu.com/ubuntu bionic/universe amd64 Packages
        100 /var/lib/dpkg/status
     4.0.0-1~bionic 500
        500 http://packages.osrfoundation.org/gazebo/ubuntu-stable bionic/main amd64 Packages

```

So not really sure what is going on, or if there's something weird with the minor releases of libignition-math. But we should fix this before the next release.

Moreover, I meant a way to check the version in the code, there's no  `#ifdef GAZEBO_MAJOR_VERSION` for libginition math that I could find...

xdwgood, 2020-05-07:
@Jaeyoung-Lim 
```dw@dw:~$ apt-cache policy libignition-math4
libignition-math4:
  Installed: 4.0.0-1~xenial
  Candidate: 4.0.0-1~xenial
  Version table:
 *** 4.0.0-1~xenial 500
        500 http://packages.osrfoundation.org/gazebo/ubuntu-stable xenial/main amd64 Packages
        100 /var/lib/dpkg/status
```

xdwgood, 2020-05-18:
fix here： https://github.com/PX4/sitl_gazebo/pull/492

closed 2020-05-18 by xdwgood