PX4/PX4-SITL_gazebo-classic #460: iris_irlock lidar ray collides with iris hitbox

artifactz, 2020-04-16:
The distance sensor of `iris_irlock` returns the same value at all times in gazebo9, because it collides with the `base_link_inertia_collision` box defined in `iris.sdf`.

Maybe collisions from within were ignored in a previous gazebo version.

A possible fix could involve refining the iris collision geometry (which is 1 box and 4 cylinders at the moment).

Jaeyoung-Lim, 2020-04-21:
@artifactz Could you maybe create a PR for the fix? You can also try changing the `self_collide` option: http://sdformat.org/spec?ver=1.7&elem=model#model_self_collide

artifactz, 2020-04-21:
All-clear. I've been using the `libgazebo_ros_range.so` plugin instead of `libgazebo_lidar_plugin.so` by accident (because I also had a model called "lidar" in my models folder).
Using `libgazebo_lidar_plugin.so`, it works as intended. Sorry for the disturbance.

closed 2020-04-21 by artifactz