PX4/PX4-SITL_gazebo-classic #464: Wrong #if on usv plugin

Pedro-Roque, 2020-04-21:
This line is wrong:
https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_usv_dynamics_plugin.cpp#L250
 #if GAZEBO_MAJOR_VERSION >= 8

I have Gazebo 9.13 and latest libignition's , and it won't build. This should instead be:

 #if GAZEBO_MAJOR_VERSION < 8

@Jaeyoung-Lim

Jaeyoung-Lim, 2020-04-21:
@Pedro-Roque Would you mind creating a fix for it?

Pedro-Roque, 2020-04-21:
Have it, will do a PR soon. Just wanted to make sure

Jaeyoung-Lim, 2020-04-21:
@Pedro-Roque One thing I don't get... I am running `9.12` and it works for me. This means there is also something else?

Pedro-Roque, 2020-04-21:
I have 4.0.0

Jaeyoung-Lim, 2020-04-21:
@Pedro-Roque Which package?

Pedro-Roque, 2020-04-21:
ignition math. Hmm, in that case, then the if should contain a mention to ignition math lib version?

Pedro-Roque, 2020-04-21:
I'd do 
`#if IGNITION_MATH_MAJOR_VERSION < 4`

instead of
`#if GAZEBO_MAJOR_VERSION >= 8`

Jaeyoung-Lim, 2020-04-21:
I have `libignition-math4/bionic,now 4.0.0+dfsg1-4 amd64 [installed,automatic]`

Pedro-Roque, 2020-04-21:
mine was installed from source, following gazebos source install. That is definitely wierd. Let me triple check that.

Pedro-Roque, 2020-04-21:
Mine is definitely 4 too. That's I'm not sure what to do now...

Jaeyoung-Lim, 2020-04-21:
@Pedro-Roque You need to check what is actually being used to compile the plugin, not the one that is installed in your system

Pedro-Roque, 2020-04-21:
I'm linking with 
`-isystem /usr/local/include/ignition/math4`

which contains in ignition/math/config.hh the following:
```
/* Config.hh. Generated by CMake for ignition-math. */

/* Version number */
#define IGNITION_MATH_MAJOR_VERSION 
#define IGNITION_MATH_MINOR_VERSION 
#define IGNITION_MATH_PATCH_VERSION 

#define IGNITION_MATH_VERSION "4.0"
#define IGNITION_MATH_VERSION_FULL "4.0.0"

#define IGNITION_MATH_VERSION_NAMESPACE v4

#define IGNITION_MATH_VERSION_HEADER "Ignition math, version 4.0.0\nCopyright (C) 2014 Open Source Robotics Foundation.\nReleased under the Apache 2.0 License.\n\n"

/* #undef IGNITION_MATH_BUILD_TYPE_PROFILE */
/* #undef IGNITION_MATH_BUILD_TYPE_DEBUG */
/* #undef IGNITION_MATH_BUILD_TYPE_RELEASE */
```

Jaeyoung-Lim, 2020-09-28:
@Pedro-Roque Is this still an issue?

Pedro-Roque, 2020-09-29:
@Jaeyoung-Lim I haven't checked this lately. I will close it for now.

closed 2020-09-29 by Pedro-Roque