PX4/PX4-SITL_gazebo-classic #477: Error while trying to reboot from terminal

TadielloM, 2020-04-29:
Hi everyone,
I'm developing an environment for Reinforcement Learning with PX4, ROS and Gazebo. To do that I would like to reboot the px4 in order to reinitialize the ekf and all the simulated sensors. 
However when I try to run reboot on the terminal where I launched ```roslaunch px4 posix_sitl.launch``` I get the error 
`ERROR [reboot] reboot failed (-22)
Command 'reboot' failed, returned -1`
What is the reason for that? and in case where can I look to try to solve this problem?

Otherwise, I was thinking to delete the model in Gazebo and reload everything from scratch, however if I'm trying in this way launching the service ```rosservice call /gazebo/delete_model/ ...```
the model is deleted but the sitl process doesn't exit properly. I was trying to search where the node sitl is created but I didn't found it and neither how it is managed in the code. 

Concluding, there is a way to start and end the sitl process without exiting from Gazebo and only reinitialize the autopilot and in case what is the best way to do that?

Jaeyoung-Lim, 2020-04-29:
@TadielloM PX4 SITL is not running inside gazebo nor ROS. Therefore, if you reboot the firmware you are just killing the process therefore the program exists. what you need to do is maybe reinitialize the ekf separately. However, this does not take your firmware to the initial state. There are a lot more "states" in the firmware than just the EKF.

Reinforcement learning against SITL is in general not a good idea
- Firmware has a lot of various states inside the software in addition to the actual dynamics "states". This makes it harder to learn since all the internal states are not observable.
- It is temporarily highly coupled due to the previous reasons, which include preflight checks, arming .etc
- You have a limit on speeding up the simulation due to the high rate of imu readings inside the firmwawre
- There are a lot of parameters that differ between vehicles. If you change just one parameter, your learning against SITL is not good anymore since the firmware behaves differently. 

I think a similar use case was needed in something like https://github.com/PX4/sitl_gazebo/pull/380, but as you can see in the comments the EKF takes quite some time to recover after you have set the vehicle in a random state

Jaeyoung-Lim, 2020-10-05:
@tadteo Any updates?

Jaeyoung-Lim, 2020-11-19:
Closing since the author didn't reply

closed 2020-11-19 by Jaeyoung-Lim

unravelwool, 2021-04-22:
Hi，I have met the same problem as you too. Could you tell me how to "reinitialize the ekf and other components separately". Is there any command in px4 terminal that can reinitialize the ekf? I did not find it. Or can I control the reinitialization of ekf by mavros   topic or gazebo plugin api?

Jaeyoung-Lim, 2021-04-22:
@unravelwool At this point, I don't think this is a supported feature