PX4/PX4-SITL_gazebo-classic #478: Error trying to start sitl_gazebo simulation

shrit, 2020-04-30:
Hello,
I have moved to gazebo 11, and since then, I am trying to start the simulation of several quadrotors using the `gazebo_sitl_multiple_run.sh` provided inside the` Tools/` directory.
The compilation process is fine the error is started when trying to spawn the second quadrotor:
The problem seems to be related to mavlink plugin.
I have tried several versions of sitl_gazebo and px4, the issue has araised only when using gazebo 11.
```
Starting gazebo
Gazebo multi-robot simulator, version 11.0.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 129.175.4.102
starting instance 0 in /meta/tansa/lib/Firmware/build/px4_sitl_default/instance_0

Spawning iris_0
starting instance 1 in /meta/tansa/lib/Firmware/build/px4_sitl_default/instance_1

Spawning iris_1
[Msg] Connecting to PX4 SITL using TCP
[Msg] Lockstep is enabled
[Msg] Speed factor set to: 1
[Err] [Model.cc:1112] Exception occured in the Load function of plugin with name[mavlink_interface] and filename[libgazebo_mavlink_interface.so]. This plugin will not run.
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
^CAn instance of Gazebo is not running.
starting instance 2 in /meta/tansa/lib/Firmware/build/px4_sitl_default/instance_2
^CTraceback (most recent call last):
  File "/meta/tansa/lib/Firmware/Tools/../Tools/sitl_gazebo/scripts/xacro.py", line 707, in main
    doc = parse(f)
  File "/usr/lib/python3.8/xml/dom/minidom.py", line 1958, in parse
    from xml.dom import expatbuilder
  File "<frozen importlib._bootstrap>", line 991, in _find_and_load
  File "<frozen importlib._bootstrap>", line 975, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 671, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 779, in exec_module
  File "<frozen importlib._bootstrap_external>", line 911, in get_code
  File "<frozen importlib._bootstrap_external>", line 580, in _compile_bytecode
KeyboardInterrupt

During handling of the above exception, another exception occurred:
```
Best regards,

Jaeyoung-Lim, 2020-04-30:
@shrit gazebo11 is not supported. The latest supported version is gazebo 9.12

shrit, 2020-04-30:
@Jaeyoung-Lim Thanks for your answer,
Unfortunately, it will be very complicated for me to go back to version 9, (I was on version 10 for more than 6 months, and last time I used version 9 was last summer) I am on Arch Linux, Tell me what is required to support version 11 and I can start a pull request.
Thanks,

Jaeyoung-Lim, 2020-04-30:
@shrit You would need to fix the exact error message you see in the log. 
```
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::event::Connection; typename boost::detail::sp_member_access<T>::type = gazebo::event::Connection*]: Assertion `px != 0' failed.
```

shrit, 2020-04-30:
@Jaeyoung-Lim This is the line that is throwing the exception: 
https://github.com/PX4/sitl_gazebo/blob/0823f5e006011d36858c2dd39881874c1edac10b/src/gazebo_mavlink_interface.cpp#L311
I do not understand the reason behind it so far,  @julianoes Do you have some thoughts one it?

Jaeyoung-Lim, 2020-05-01:
@shrit Most likely is is coming from 

https://github.com/PX4/sitl_gazebo/blob/0823f5e006011d36858c2dd39881874c1edac10b/src/gazebo_mavlink_interface.cpp#L310

And maybe it is not able to find what is trying to find.

shrit, 2020-05-01:
@Jaeyoung-Lim Actually, I have checked this function, it is returing `true`.

julianoes, 2020-05-04:
@shrit can you try to start add `gdb -ex run` before `gzserver` to get a backtrace of what is happening?

shrit, 2020-05-04:
@julianoes Here is a full backtrace for gzserver with gdb:

```
#0  0x00007ffff1d50f25 in raise () at /usr/lib/libc.so.6
#1  0x00007ffff1d3a897 in abort () at /usr/lib/libc.so.6
#2  0x00007ffff1d3a767 in _nl_load_domain.cold () at /usr/lib/libc.so.6
#3  0x00007ffff1d49526 in  () at /usr/lib/libc.so.6
#4  0x00007fff8c0d0985 in boost::shared_ptr<gazebo::event::Connection>::operator->() const ()
    at /meta/lemon/lib/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_multirotor_base_plugin.so
#5  0x00007ffe90cde1a7 in gazebo::GazeboMavlinkInterface::~GazeboMavlinkInterface() () at /meta/lemon/lib/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so
#6  0x00007ffe90cde6ce in gazebo::GazeboMavlinkInterface::~GazeboMavlinkInterface() () at /meta/lemon/lib/Firmware/build/px4_sitl_default/build_gazebo/libgazebo_mavlink_interface.so
#7  0x00007ffff6fd67da in boost::detail::sp_counted_base::release() (this=0x7ffebc0b9a30) at /usr/include/boost/smart_ptr/detail/sp_counted_base_std_atomic.hpp:112
#8  0x00007ffff6fd67da in boost::detail::sp_counted_base::release() (this=0x7ffebc0b9a30) at /usr/include/boost/smart_ptr/detail/sp_counted_base_std_atomic.hpp:108
#9  0x00007ffff711f48f in boost::detail::shared_count::~shared_count() (this=0x7ffefbffe318, __in_chrg=<optimized out>) at /usr/include/boost/smart_ptr/detail/shared_count.hpp:427
        pluginName = "mavlink_interface"
        filename = "libgazebo_mavlink_interface.so"
        plugin = {px = 0x7ffebc0ba920, pn = {pi_ = 0x7ffebc0b9a30}}
#10 0x00007ffff711f48f in boost::shared_ptr<gazebo::ModelPlugin>::~shared_ptr() (this=0x7ffefbffe310, __in_chrg=<optimized out>) at /usr/include/boost/smart_ptr/shared_ptr.hpp:341
        pluginName = "mavlink_interface"
        filename = "libgazebo_mavlink_interface.so"
        plugin = {px = 0x7ffebc0ba920, pn = {pi_ = 0x7ffebc0b9a30}}
#11 0x00007ffff711f48f in gazebo::physics::Model::LoadPlugin(std::shared_ptr<sdf::v9::Element>) (this=<optimized out>, _sdf=...) at /usr/include/boost/smart_ptr/shared_ptr.hpp:341
        pluginName = "mavlink_interface"
        filename = "libgazebo_mavlink_interface.so"
        plugin = {px = 0x7ffebc0ba920, pn = {pi_ = 0x7ffebc0b9a30}}
#12 0x00007ffff7121400 in gazebo::physics::Model::LoadPlugins() (this=0x555556b88470) at /usr/include/c++/9.2.0/ext/atomicity.h:96
        pluginElem = std::shared_ptr<class sdf::v9::Element> (use count 3, weak count 25) = {get() = 0x555557a57820}
        iterations = <optimized out>
#13 0x00007ffff71677a4 in gazebo::physics::World::LoadPlugins() (this=0x555556e08e00) at /usr/include/boost/smart_ptr/shared_ptr.hpp:732
        model = {px = <optimized out>, pn = {pi_ = 0x555556de91e0}}
        i = 3
#14 0x00007ffff7179295 in gazebo::physics::World::Step() (this=0x555556e08e00) at /meta/gazebo/gazebo/physics/World.cc:677
        updatePeriod = <optimized out>
        tmpTime =
          {_vptr.Time = 0x0, static Zero = {_vptr.Time = 0x7ffff333b8b8 <vtable for gazebo::common::Time+16>, static Zero = <same as static member of an already seen type>, static Second = {_vptr.Time = 0x7ffff333b8b8 <vtable for gazebo::common::Time+16>, static Zero = <same as static member of an already seen type>, static Second = <same as static member of an already seen type>, static Hour = {_vptr.Time = 0x7ffff333b8b8 <vtable for gazebo::common::Time+16>, static Zero = <same as static member of an already seen type>, static Second = <same as static member of
an already seen type>, static Hour = <same as static member of an already seen type>, sec = 3600, nsec = 0, static nsInSec = 1000000000, static nsInMs = 1000000, static wallTime = {_vptr.Time = 0x7ffff333b8b8 <vtable for gazebo::common::Time+16>, static Zero = <same as static member of an already seen type>, static Second = <same as static member of an already seen type>, static Hour = <same as static member of an already seen type>, sec = 1588588978, nsec = 939716910, static nsInSec = 1000000000, static nsInMs = 1000000, static wallTime = <same as static member of an already seen type>, static wallTimeISO = {static npos = 18446744073709551615, _M_dataplus = {<std::allocator<char>> = {<__gnu_cxx::new_allocator<char>> = {<No data fields>}, <No data fields>}, _M_p = 0x555556b03840 "2020-05-04T12:42:49.913035"}, _M_string_length = 26, {_M_local_buf = "\036", '\000' <repeats 14 times>, _M_allocated_capacity = 30}}, static clockResolution =
{tv_sec = 0, tv_nsec = 1}}, static wallTimeISO = <same as static member of an already seen type>, static clockResolution = <same as static member of an already seen type>}, sec = 1, nsec = 0, static nsInSec = 1000000000, static nsInMs = 1000000, static wallTime = <same as static member of an already seen type>, static wallTimeISO = <same as static member of an already seen type>, static clockResolution = <same as static member of an already seen type>}, static Hour = <same as static member of an already seen type>, sec = 0, nsec = 0, static nsInSec = 1000000000, static nsInMs = 1000000, static wallTime = <same as static member of an already seen type>, static wallTimeISO = <same as static member of an already seen type>, static clockResolution = <same as static member of an already seen type>}, static Second = <same as static member of an already seen type>, static Hour = <same as static member of an already seen type>, sec = 0, nsec =
[... cut ...] een type>}
#15 0x00007ffff717c3d5 in gazebo::physics::World::RunLoop() (this=0x555556e08e00) at /meta/gazebo/gazebo/physics/World.cc:551
#16 0x00007ffff1fc5ee4 in std::execute_native_thread_routine(void*) (__p=0x555556efdf00) at /build/gcc/src/gcc/libstdc++-v3/src/c++11/thread.cc:80
        __t = std::unique_ptr<struct std::thread::_State> = {get() = 0x555556efdf00}
#17 0x00007ffff73b94cf in start_thread () at /usr/lib/libpthread.so.0
#18 0x00007ffff1e142d3 in clone () at /usr/lib/libc.so.6

```

julianoes, 2020-05-05:
@shrit ok looks like it crashes on destruction. So the next step is to find out why it wants to exit in the first place.

shrit, 2020-06-06:
@julianoes I think we can close this as you have fixed the issue in 4acd679d17a735a1f62781349d14289a29b01014

closed 2020-06-06 by shrit