PX4/PX4-SITL_gazebo-classic #481: `make px4_sitl gazebo` error (@ gazebo_wind_plugin)

breathedigital, 2020-05-02:
Hello -
`px4 px4_sitl gazebo` resulted in the build process being stuck at **gazebo_wind_plugin** (for almost 10-15 mins) before spitting out this error. On Ubuntu 18.x running on docker (on Mac).
 
Your thoughts are highly appreciated!

```
Scanning dependencies of target systemcmds__tune_control
[ 80%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_rc.c.o
[ 80%] Building CXX object src/systemcmds/tune_control/CMakeFiles/systemcmds__tune_control.dir/tune_control.cpp.o
[ 80%] Building CXX object src/systemcmds/topic_listener/CMakeFiles/systemcmds__topic_listener.dir/listener_generated.cpp.o
[ 80%] Building CXX object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_search_min.cpp.o
[ 80%] Linking CXX static library libsystemcmds__tune_control.a
[ 80%] Built target systemcmds__tune_control
[ 80%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_servo.c.o
[ 80%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_sleep.c.o
Scanning dependencies of target systemcmds__ver
[ 80%] Building C object src/systemcmds/ver/CMakeFiles/systemcmds__ver.dir/ver.c.o
[ 80%] Building CXX object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_smooth_z.cpp.o
[ 81%] Linking CXX static library libsystemcmds__ver.a
[ 81%] Built target systemcmds__ver
Scanning dependencies of target systemcmds__work_queue
[ 81%] Building CXX object src/systemcmds/work_queue/CMakeFiles/systemcmds__work_queue.dir/work_queue_main.cpp.o
[ 81%] Building CXX object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_tone.cpp.o
[ 81%] Linking CXX static library libsystemcmds__work_queue.a
[ 81%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_uart_baudchange.c.o
[ 81%] Built target systemcmds__work_queue
[ 81%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_uart_console.c.o
[ 59%] Built target gazebo_wind_plugin
Makefile:151: recipe for target 'all' failed
make[5]: *** [all] Error 2
platforms/posix/CMakeFiles/sitl_gazebo.dir/build.make:110: recipe for target 'external/Stamp/sitl_gazebo/sitl_gazebo-build' failed
make[4]: *** [external/Stamp/sitl_gazebo/sitl_gazebo-build] Error 2
CMakeFiles/Makefile2:39323: recipe for target 'platforms/posix/CMakeFiles/sitl_gazebo.dir/all' failed
make[3]: *** [platforms/posix/CMakeFiles/sitl_gazebo.dir/all] Error 2
make[3]: *** Waiting for unfinished jobs....
[ 81%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_uart_send.c.o
[ 82%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_uart_loopback.c.o
[ 82%] Linking CXX static library libsystemcmds__topic_listener.a
[ 82%] Building CXX object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/test_versioning.cpp.o
[ 82%] Building C object src/systemcmds/tests/CMakeFiles/systemcmds__tests.dir/tests_main.c.o
[ 82%] Built target systemcmds__topic_listener
[ 82%] Linking CXX static library libsystemcmds__tests.a
[ 82%] Built target systemcmds__tests
CMakeFiles/Makefile2:20407: recipe for target 'platforms/posix/CMakeFiles/gazebo.dir/rule' failed
make[2]: *** [platforms/posix/CMakeFiles/gazebo.dir/rule] Error 2
Makefile:5724: recipe for target 'gazebo' failed
make[1]: *** [gazebo] Error 2
Makefile:205: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 2
root@a2cfc09fde1c:/PX4/Firmware# 
```

Jaeyoung-Lim, 2020-05-09:
@breathedigital Any updates?

Jaeyoung-Lim, 2020-09-28:
Closing since the issue is stale.

Please feel free to reopen if the issue presists

closed 2020-09-28 by Jaeyoung-Lim