PX4/PX4-SITL_gazebo-classic #486: Aliasing on motors might occur error

shrit, 2020-05-05:
Hello,

I am trying to add three links inside the iris quadrotor model, each link has a cylinder form with correct inertia in place. Each cylinder represents a senors, two inertial are placed on quadrotors arms (which connect the motors with the body) and the third is placed in the center of the quadrotor (like an opt-flow camera).
Before adding the inertia values, simulation started fine without any error, but when trying to take off the quadrotors, it went randomly in all directions to end by a crash. Then, I figured out that it might be related to the inertia value, so I calculated these values. Now, when I try to start the simulation I get this error printed from all quadrotors: 
https://github.com/PX4/sitl_gazebo/blob/2736a9134d7fed0cae7b7836d4370f9a4dfef439/src/gazebo_motor_model.cpp#L194

Do you have any idea why this error is occurring?
Best regards,

Jaeyoung-Lim, 2020-05-05:
@shrit It usually happens when you have a unstable/diverging joints due to conflicting constraints.

shrit, 2020-05-05:
@Jaeyoung-Lim Do I have to have the same mass applied to four arms instead of two arms? adding  2 more sensors to make everything equal? (Knowing that I am adding only 10 grams on two arms)

Jaeyoung-Lim, 2020-05-05:
@shrit Not really. It is more related to the joints you are using. If you are using a fixed joint, it might be causing issues

shrit, 2020-05-05:
@Jaeyoung-Lim Thanks for replay.
Exactly I am using a fixed joint, Which one can I use (since the sensor only needs to be fixed,  revolute maybe? but I am not sure of configurations)

Jaeyoung-Lim, 2020-08-12:
@shrit Any updates?

shrit, 2020-08-12:
Yes, sorry for the late response, I have changed the joint types, I am using revolute joint now and it is working perfectly.
I think we can close this issue now.

Jaeyoung-Lim, 2020-08-12:
@shrit Thanks for the update

closed 2020-08-12 by Jaeyoung-Lim

shrit, 2020-11-13:
@Jaeyoung-Lim I have just updated my repo to the master. Now I am facing similar issue, when adding these sensors, the quadrotors are jumping arounds, even though I am using the revolute joint.
Is there any change to the code base? or anything that might bring this issue up,  I am using the same version of gazebo

Jaeyoung-Lim, 2020-11-13:
@shrit I am not aware what your previous version was, so I cannot say.

If you have issues please post a issue with a concise description and logs if possible

shrit, 2020-11-13:
I do not have any log, I am trying to add sensors to the `iris.sdf`
I used always master, so I do not have any specific version, I know that there was not a gps sensors, and the iris sdf were generated using `urdf`
When adding the sensor to the quadrotors, they jump and reverse, with non sense behavior.
The last time I used `revolute` joint instead of `fixed` to resolve the issue. Now it is back with `revolute`.
If I try to use `fixed` joint I have only one sensor that is available the others can not be seen or found, even I tried to follow them, but at least the quadrotors do not jump.