PX4/PX4-SITL_gazebo-classic #487: error while excuting "make px4_sitl gazebo" on macOS

kainism, 2020-05-07:
”make px4_sitl gazebo“ fails on information as follows.

- OS: macOS Catalina 10.15.4

- gazebo version: 9.13

- opencv version: 4.3.0

`kevin@kevinJdeMacBook-Pro ~ % cd /Users/kevin/Firmware
kevin@kevinJdeMacBook-Pro Firmware % make px4_sitl gazebo
[0/7] Performing build step for 'sitl_gazebo'
[5/73] Generating /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
FAILED: /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf 
cd /Users/kevin/Firmware/Tools/sitl_gazebo && rm -f /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && /usr/local/bin/python3 /Users/kevin/Firmware/Tools/sitl_gazebo/scripts/xacro.py -o /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro enable_mavlink_interface:=true enable_ground_truth:=false enable_wind:=false enable_logging:=false rotors_description_dir:=/Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description send_vision_estimation:=true send_odometry:=false && gz sdf -p /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && rm -f /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf

dyld: Library not loaded: /usr/local/opt/bullet/lib/libBulletSoftBody.2.88.dylib
  Referenced from: /usr/local/bin/gz
  Reason: image not found
/bin/sh: line 1:  6646 Abort trap: 6           gz sdf -p /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[18/73] Building CXX object CMakeFiles...dir/src/gazebo_gst_camera_plugin.cpp.o
ninja: build stopped: subcommand failed.
[3/7] Linking CXX executable bin/px4
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/kevin/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1`

Thanks a lot!

kainism, 2020-05-09:
I just run "brew cleanup" before "make px4_sitl gazebo", and the error info changes as shown below:

kevin@kevinJdeMacBook-Pro Firmware % make px4_sitl gazebo
[0/4] Performing build step for 'sitl_gazebo'
[2/44] Generating /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
FAILED: /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf 
cd /Users/kevin/Firmware/Tools/sitl_gazebo && rm -f /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && /usr/local/bin/python3 /Users/kevin/Firmware/Tools/sitl_gazebo/scripts/xacro.py -o /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.xacro enable_mavlink_interface:=true enable_ground_truth:=false enable_wind:=false enable_logging:=false rotors_description_dir:=/Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description send_vision_estimation:=true send_odometry:=false && gz sdf -p /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf && rm -f /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf

dyld: Library not loaded: /usr/local/opt/bullet/lib/libBulletSoftBody.2.88.dylib
  Referenced from: /usr/local/bin/gz
  Reason: image not found
/bin/sh: line 1: 73710 Abort trap: 6           gz sdf -p /Users/kevin/Firmware/Tools/sitl_gazebo/models/rotors_description/urdf/iris_base.urdf >> /Users/kevin/Firmware/Tools/sitl_gazebo/models/iris/iris.sdf
[15/44] Building CXX object CMakeFiles/gazebo_geotagged_images_plugin.dir/src/gazebo_geotagged_images_plugin.cpp.o
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/kevin/Firmware/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.17.2/bin/cmake --build .
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

Jaeyoung-Lim, 2020-05-09:
Related to https://github.com/osrf/homebrew-simulation/issues/1003

Jaeyoung-Lim, 2020-10-05:
Closing since this is a duplicate of https://github.com/PX4/sitl_gazebo/issues/513

closed 2020-10-05 by Jaeyoung-Lim