PX4/PX4-SITL_gazebo-classic #489: Unable to launch iris_fpv_cam, iris_depth_camera and others

Rezenders, 2020-05-07:
Hello,

I am trying to run a mavros + gazebo simulation with an iris model with a camera, but I am not able to do so.
 
When I run a simulation with the following command everything works just fine (this model does not have a camera):
```
$ roslaunch px4 mavros_posix_sitl.launch 
```
However, when I run a simulation with any of the models  `iris_fpv_cam`, `iris_depth_camera`, `iris_downward_depth_camera`, or `iris_stereo_camera` it does not work, I am running the following command:

```
$ roslaunch px4 mavros_posix_sitl.launch vehicle:=iris_fpv_cam
```

As mentioned in the issue #228 I tried to create a launch file changing the vehicle  and sdf arg, but it did not work:

```
<?xml version="1.0"?>
<launch>
    <arg name="vehicle" default="iris_depth_camera"/>
    <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>

    <include file="$(find px4)/launch/mavros_posix_sitl.launch">
        <arg name="respawn_gazebo" value="true"/>
        <arg name="respawn_mavros" value="true"/>
        <arg name="vehicle" value="$(arg vehicle)"/>
        <arg name="sdf" value="$(arg sdf)"/>
        <arg name="verbose" value="true"/>
    </include>
</launch>
```
When building the Firmware repo I followed the instructions on https://dev.px4.io/v1.9.0/en/simulation/multi-vehicle-simulation.html:

```
$ cd Firmware_clone
$ git submodule update --init --recursive
$ DONT_RUN=1 make px4_sitl_default gazebo
$ source Tools/setup_gazebo.bash $(pwd) $(pwd)/build/px4_sitl_default
$ export ROS_PACKAGE_PATH=$ROS_PACKAGE_PATH:$(pwd):$(pwd)/Tools/sitl_gazebo
```

I found a similar question on ros answers: https://answers.ros.org/question/345462/trouble-running-gazebo-simulation-with-camera-plugin/

mzwtju, 2020-05-09:
https://github.com/PX4/sitl_gazebo/issues/228 try this

mzwtju, 2020-05-11:
The launch file you created seem to be imcomplete
In `mavros_posix_sitl.launch `
add  `<arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris_fpv_cam/iris_fpv_cam.sdf"/>`
It should work.

Rezenders, 2020-05-13:
> The launch file you created seem to be imcomplete
> In `mavros_posix_sitl.launch `
> add `<arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris_fpv_cam/iris_fpv_cam.sdf"/>`
> It should work.

I get the same error when I do that. Maybe it is a problem in the installation and sourcing?

closed 2020-05-13 by Rezenders

induchandran77, 2023-04-07:
this works for single vehicle. How can I do it for multiple vehicles?
I use multi_uav_mavros_sitl.launch
<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
            <arg name="gst_udp_port" value="$(eval 5600 + arg('ID'))"/>
            <arg name="video_uri" value="$(eval 5600 + arg('ID'))"/>
            <arg name="mavlink_cam_udp_port" value="$(eval 14530 + arg('ID'))"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>

induchandran77, 2023-04-07:
<?xml version="1.0"?>
<launch>
    
    
    
    <arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="world" default="$(find mavlink_sitl_gazebo)/worlds/empty.world"/>
    
    <arg name="gui" default="true"/>
    <arg name="debug" default="false"/>
    <arg name="verbose" default="false"/>
    <arg name="paused" default="false"/>
    
    <include file="$(find gazebo_ros)/launch/empty_world.launch">
        <arg name="gui" value="$(arg gui)"/>
        <arg name="world_name" value="$(arg world)"/>
        <arg name="debug" value="$(arg debug)"/>
        <arg name="verbose" value="$(arg verbose)"/>
        <arg name="paused" value="$(arg paused)"/>
    </include>
    
    <group ns="uav0">
        
        <arg name="ID" value="0"/>
        <arg name="fcu_url" default="udp://:14540@localhost:14580"/>
        
        <include file="$(find px4)/launch/single_vehicle_spawn.launch">
            <arg name="x" value="0"/>
            <arg name="y" value="0"/>
            <arg name="z" value="0"/>
            <arg name="R" value="0"/>
            <arg name="P" value="0"/>
            <arg name="Y" value="0"/>
            <arg name="vehicle" value="$(arg vehicle)"/>
            <arg name="mavlink_udp_port" value="14560"/>
            <arg name="mavlink_tcp_port" value="4560"/>
            <arg name="ID" value="$(arg ID)"/>
            <arg name="gst_udp_port" value="$(eval 5600 + arg('ID'))"/>
            <arg name="video_uri" value="$(eval 5600 + arg('ID'))"/>
            <arg name="mavlink_cam_udp_port" value="$(eval 14530 + arg('ID'))"/>
        </include>
        
        <include file="$(find mavros)/launch/px4.launch">
            <arg name="fcu_url" value="$(arg fcu_url)"/>
            <arg name="gcs_url" value=""/>
            <arg name="tgt_system" value="$(eval 1 + arg('ID'))"/>
            <arg name="tgt_component" value="1"/>
        </include>
    </group>

induchandran77, 2023-04-07:
<arg name="est" default="ekf2"/>
    <arg name="vehicle" default="iris"/>
    <arg name="ID" default="1"/>
    <env name="PX4_SIM_MODEL" value="$(arg vehicle)" />
    <arg name="mavlink_udp_port" default="14560"/>
    <arg name="mavlink_tcp_port" default="4560"/>
    <arg name="gst_udp_port" default="5600"/>
    <arg name="video_uri" default="5600"/>
    <arg name="mavlink_cam_udp_port" default="14530"/>
    <arg name="mavlink_id" value="$(eval 1 + arg('ID'))" />
    
    <arg name="interactive" default="true"/>
    
    <arg name="cmd" default="$(find mavlink_sitl_gazebo)/scripts/jinja_gen.py --stdout --mavlink_id=$(arg mavlink_id) --mavlink_udp_port=$(arg mavlink_udp_port) --mavlink_tcp_port=$(arg mavlink_tcp_port) --gst_udp_port=$(arg gst_udp_port) --video_uri=$(arg video_uri) --mavlink_cam_udp_port=$(arg mavlink_cam_udp_port) $(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf.jinja $(find mavlink_sitl_gazebo)"/>
    <param command="$(arg cmd)" name="sdf_$(arg vehicle)$(arg ID)"/>
    
    <arg unless="$(arg interactive)" name="px4_command_arg1" value=""/>
    <arg     if="$(arg interactive)" name="px4_command_arg1" value="-d"/>
    <node name="sitl_$(arg ID)" pkg="px4" type="px4" output="screen" args="$(find px4)/build/px4_sitl_default/etc -s etc/init.d-posix/rcS -i $(arg ID) -w sitl_$(arg vehicle)_$(arg ID) $(arg px4_command_arg1)">
    </node>
    
    <node name="$(anon vehicle_spawn)" pkg="gazebo_ros" type="spawn_model" output="screen" args="-sdf -param sdf_$(arg vehicle)$(arg ID) -model $(arg vehicle)$(arg ID) -x $(arg x) -y $(arg y) -z $(arg z) -R $(arg R) -P $(arg P) -Y $(arg Y)"/>
</launch>

lijoe123, 2023-04-10:
Which verson you use? Noetic or melodic? @induchandran77

induchandran77, 2023-04-10:
I use Noetic

lijoe123, 2023-04-10:
[image]
[image]
So if in one vehicle, I only had the two topic of camera. And the realsense in the iris is weird. What about you? @induchandran77

nicolasfripp, 2023-06-09:
> this works for single vehicle. How can I do it for multiple vehicles?
> I use multi_uav_mavros_sitl.launch

Could you find a solution? @induchandran77

hao158, 2024-10-26:
> [image] [image] So if in one vehicle, I only had the two topic of camera. And the realsense in the iris is weird. What about you? @induchandran77

I have the same difficulty, have you solved it