PX4/PX4-SITL_gazebo-classic #491: how to enable cameras in multi-uav simulation

mzwtju, 2020-05-09:
My PX4 firmware version is v1.9.2

I followed the instruction in https://github.com/PX4/sitl_gazebo/issues/228
I changed directly the `sdf` argument.
` <arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/iris_fpv_cam/iris_fpv_cam.sdf"/>` in the `mavros_posix_sitl.launch ` file  and it workd well

Now I'm trying to enable camera in every drones I launch in the Gazebo when I'm using
`multi_uav_mavros_sitl.launch` file.

I tried many ways like adding the `sdf` arg in the `single_vehicle_spawn.launch `file 
but it did not work out.

I'm wondering how to do this?

thanks in advance!

Jaeyoung-Lim, 2020-05-09:
@mzwtju which arg were you changing?

mzwtju, 2020-05-09:
> @mzwtju which arg were you changing?

actually adding a  `sdf`  arg 

in https://github.com/PX4/sitl_gazebo/issues/228

> So you need to keep the `name` argument as `iris`, and change directly the `sdf` argument.
>`arg name="sdf" value="$(find mavlink_sitl_gazebo)/models/$(arg my_model_name)/$(arg my_model_name).sdf"/`
>

I'm using `multi_uav_mavros_sitl.launch` file to launch multi-uav

So I take this https://github.com/PX4/sitl_gazebo/issues/228 as an example and change  `single_vehicle_spawn.launch`  file by adding a `sdf` arg

Jaeyoung-Lim, 2020-05-09:
@mzwtju It should work that way. How were you changing them? Do you have a different sdf file for each vehicle?

mzwtju, 2020-05-11:
@Jaeyoung-Lim  Sorry for my late reply

This is how I do it:

1. I copy the `iris_fpv_cam` fie in the `Tools/sitl_gazebo/models` and create a new file named `iris_fpv_cam2`
2. changed the` iris_fpv_cam2.sdf` file and `model.config` file as well
`<model name='iris_fpv_cam2'>`
```
    <include>
      <uri>model://fpv_cam2</uri>
      <pose>0 0 0 0 0 0</pose>
    </include>
```
[image]
3. In the `multi_uav_mavros_sitl.launch `file 
 I added `<arg name="sdf" value="$(find mavlink_sitl_gazebo)/models/iris_fpv_cam2/iris_fpv_cam2.sdf"/>` and
`<arg name="sdf" value="$(find mavlink_sitl_gazebo)/models/iris_fpv_cam/iris_fpv_cam.sdf"/>`  to the 2 drones  I want to launch
4. in the `single_vehicle_spawn.launch` file
added `<arg name="sdf" default="$(find mavlink_sitl_gazebo)/models/$(arg vehicle)/$(arg vehicle).sdf"/>`

It did not work that way. Do you know what part I'm missing?

Jaeyoung-Lim, 2020-05-11:
@mzwtju You need to modify each model to use different [mavlink ports](https://github.com/PX4/sitl_gazebo/blob/master/models/rotors_description/urdf/iris_base.xacro#L10) as well as the fpv cam to use different [udp ports](https://github.com/PX4/sitl_gazebo/blob/master/models/fpv_cam/fpv_cam.sdf#L91).

Alternatively, have you checked the instructions in http://dev.px4.io/master/en/simulation/multi-vehicle-simulation.html#no_ros ? Currently this is automated by xacro macros with a bashscript without ros.

mzwtju, 2020-05-11:
@Jaeyoung-Lim  
Thanks for your help.
I didn't found [udp ports](https://github.com/PX4/sitl_gazebo/blob/master/models/fpv_cam/fpv_cam.sdf#L91) in my fpv_cam.file .

I insert udp part and modify different ports.
```
          <plugin name="GstCameraPlugin" filename="libgazebo_gst_camera_plugin.so">
            <robotNamespace></robotNamespace>
            <udpPort>5600</udpPort>
          </plugin>
```
Still can't work it out.
Do you have any example? I'm a newbie in this.

Jaeyoung-Lim, 2020-05-11:
@mzwtju I already sent you the example where you can do it "without" ROS. If you want to indeed use ros, you need to figure out what is wrong e.g. launching different models separately to see if it works

mzwtju, 2020-05-11:
@Jaeyoung-Lim 
I want to use ROS to do this.
Appreciate your help! I would figure out what I'm missing

Jaeyoung-Lim, 2020-05-12:
@mzwtju Did you figure it out?

mzwtju, 2020-05-12:
@Jaeyoung-Lim  Nope.

I think I could use newer version after v1.10.x because there is a new `single_vehicle_spawn_sdf.launch` file 
Here's a example [https://gitee.com/robin_shaun/XTDrone/blob/master/coordination/launch/multi_vehicle.launch#L152](url)

Is this https://github.com/PX4/Firmware/issues/14323 fixed? When I update submodule, same error occur :
```
fatal: reference is not a tree: 8569aec5bb709acb2fec7ccb5a5a4405f0cdec2a
Unable to checkout '8569aec5bb709acb2fec7ccb5a5a4405f0cdec2a' in submodule path 'Tools/sitl_gazebo'
```
Tried run `git submodule sync` or  `make submodulesclean`. It won't work.

Jaeyoung-Lim, 2020-05-12:
@mzwtju Yes, it is fixed. I don't quite understand. Were you trying to use multivehicle sim with XTDrone? I have never looked at the project, so I am not sure how it works

mzwtju, 2020-05-12:
@Jaeyoung-Lim 
Yes. I want to use XTDrone multivehicle sim.
It seems by using `xmlstarlet` tools  to change UDP ports in SDF file. I am not sure either.

mzwtju, 2020-05-12:
@Jaeyoung-Lim 
How to fix PX4/Firmware#14323 btw.XD

Jaeyoung-Lim, 2020-05-12:
@mzwtju It is fixed as I already stated in the comment above

mzwtju, 2020-05-12:
@Jaeyoung-Lim  OK. Thank you.

closed 2020-05-15 by mzwtju

adbidwai, 2021-07-07:
@mzwtju Were you able to run many drones with camera feeds in your multi UAV simulation using ROS and Gazebo?

mzwtju, 2021-07-07:
Yes. But I don't remember the specific steps to run that simulation. I remember I took the XTDrone project for example. 
https://gitee.com/robin_shaun/XTDrone#/robin_shaun/XTDrone/blob/master/README.md 
https://www.yuque.com/xtdrone/manual_en @adbidwai