PX4/PX4-SITL_gazebo-classic #493: How to control the gimbal on the typhoon_h480.sdf?

petertheprocess, 2020-05-14:
Hi everyone!

I followed the example in
https://dev.px4.io/v1.9.0/en/advanced/gimbal_control.html
```
vmount test yaw 30
```
then I get
```
pxh> vmount test yaw 30
WARN  [vmount] mount driver already running, run vmount stop before 'vmount test'
Command 'vmount' failed, returned 1.
```
so I run  vmount stop at first, and try again
```
pxh> vmount stop
pxh> vmount test yaw 30
INFO  [vmount] Starting in test mode
```
it seems that the command did work, but the gimbal model in gazebo did not move.

>to control the gimbal. Note that the simulated gimbal stabilizes itself, so if you send mavlink commands, set the stabilize flags to false.
Maybe  this is the reason？ But how to set the stabilize flags to false？

And I also want to konw, can I use mavros to control the gimbal? I find these topics/service in mavros, 
```
# topic
/mavros/mount_control/command
/mavros/mount_control/orientation
# service
/mavros/mount_control/configure

```
but when I try to pub a msg like this
`rostopic pub /mavros/mount_control/command mavros_msgs/MountControl "header:
  seq: 0
  stamp: {secs: 0, nsecs: 0}
  frame_id: ''
mode: 0
pitch: 0.0
roll: 1.0
yaw: 1.0
altitude: 0.0
latitude: 0.0
longitude: 0.0"
 `
also, nothing happen on gazebo model.

Jaeyoung-Lim, 2020-05-14:
@petertheprocess Here is an example for you: https://github.com/Jaeyoung-Lim/mavros_humantracking

petertheprocess, 2020-05-14:
> @petertheprocess Here is an example for you: https://github.com/Jaeyoung-Lim/mavros_humantracking
Hi, thanks for your example. But I sitl have some confusion. 
I find in your demo, `/mavros/actuator_control` which contains the pitch & yaw target angle  is published, so why we need  "/mavros/mount_control/command"? What role do these two msg play?

Jaeyoung-Lim, 2020-05-15:
@petertheprocess You can use either. you can either use actuator controls or mount control messages.

petertheprocess, 2020-05-16:
> @petertheprocess You can use either. you can either use actuator controls or mount control messages.

thx a lot, I got it. But it seems not support the lock-step SITL. What a pity. 😭

closed 2020-05-16 by petertheprocess

Jaeyoung-Lim, 2020-05-16:
@petertheprocess Why would uou want to use actuator setpoints? Why not just use mount control messages?

petertheprocess, 2020-05-18:
> @petertheprocess Why would uou want to use actuator setpoints? Why not just use mount control messages?
Oh, I naturally thought that neither of these supports lock-step SITL.

Now, when I send a mount control messages, I got this, and the gimbal did not move. 
```
[ WARN] [1589781423.872144431, 431.152000000]: CMD: Unexpected command 205, result 0
```
What could be the reason？

Jaeyoung-Lim, 2020-05-18:
@petertheprocess I don't think that is related. Did you try running the example package? Does it not work?