PX4/PX4-SITL_gazebo-classic #498: [gazebo_mavlink_interface.cpp] bind failed when specifying ip address.

jeprince, 2020-05-19:
I cloned PX4/Firmware, built stil_gazebo. `make px4_sitl gazebo ` works fine, and gazebo loads with no problems. If i modify /sitl_gazebo/models/iris/iris.sdf, and change `<mavlink_addr>INADDR_ADDR</mavlink_addr>` to `<mavlink_addr>172.20.0.1</mavlink_addr>`  (or 127.0.0.1 or any ip any docker containers running PX4 firmware) I get:

`[Err] [gazebo_mavlink_interface.cpp:561] bind failed: Cannot assign requested address, aborting`

Jaeyoung-Lim, 2020-05-21:
@jeprince  If you are running SITL in a docker container, you need to either expose the ports manually or run it in network "host" mode

jeprince, 2020-05-21:
My docker containers are using a bridge network. I believe, if i use "host" I don't have the ability to manually assign IP addresses and only one container could bind to 4560 (instead of using a bridge network with multiple IPs listening on 4560 for gazebo)

Jaeyoung-Lim, 2020-05-21:
@jeprince Okay, so either way I believe this is the issue of your contianer configuration and not a issue of this code base.

jeprince, 2020-05-21:
@Jaeyoung-Lim For me, gazebo works fine with 5+ SITL PX4s, with different IPs on UDP 14560. If they were on the "Host" network would only allow one SITL PX4 bind to localhost:4560, I can ping all containers from the host so this leads me to believe it is a problem with a gazebo plugin.

Jaeyoung-Lim, 2020-08-12:
@jeprince If you want to run multiple instances of your container, you need to assign different ports to each container

You can look at https://github.com/PX4/Firmware/blob/master/Tools/gazebo_sitl_multiple_run.sh for inspiration

Jaeyoung-Lim, 2020-10-05:
@jeprince Any updates?

closed 2020-12-17 by Jaeyoung-Lim