PX4/PX4-SITL_gazebo-classic #499: CMake Error: _MAVLINK_INCLUDE_DIR

daylanc, 2020-05-20:
Hello,

I've ran into the following error when running the `cmake ..` command:

[image]

Any possible ideas that will solve this issue?

Thanks.

Jaeyoung-Lim, 2020-05-21:
@daylanc Are you trying to build this standalone? you are missing mavlink in your system

daylanc, 2020-05-23:
@Jaeyoung-Lim I figured I didn't install the dependencies. I found my way to the following website:
https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html
and started following it. I still am having issues finding mavlink for some reason. Not sure if I'm installing correctly. Do I need to install mavlink separately?

Jaeyoung-Lim, 2020-05-23:
@daylanc You still haven't answered the question I asked. are you trying to build this repo separately? or part of px4/firmware?

daylanc, 2020-05-23:
@Jaeyoung-Lim part of px4/Firmware.

Jaeyoung-Lim, 2020-05-23:
@daylanc Why do you need to run `cmake ..` ?

Does the normal build instruction not work?

daylanc, 2020-05-24:
@Jaeyoung-Lim I'm following the following website (https://dev.px4.io/master/en/setup/dev_env_linux_ubuntu.html) to install PX4/Firmware via the ubunut.sh and ubuntu_sim_ros_melodic.sh scripts. Not sure if this is the correct way, but I can successfully run the ubuntu.sh script, but run into errors with the ubuntu_sim_ros_melodic.sh scripts.
Seems like its related to mavlink again.

Jaeyoung-Lim, 2020-05-24:
@daylanc why did you need to run `cmake ..` ?

Does 
```
make px4_sitl gazebo
```
Not work?

daylanc, 2020-05-27:
@Jaeyoung-Lim `make px4_sitl gazebo` works! Is this px4/sitl_gazebo repo used for stand-alone installation aside from the px4/Firmware?

Also, on another note, where does `make px4_sitl gazebo` grab its world files and iris quadcopter? In other words, if I wanted to modify the default world and modify/add payloads to the default quadcopter, how could I do that? Which file could I take a look at?

TSC21, 2020-05-27:
@daylanc I recommend you to have a look at the documentation before trying anything else: https://dev.px4.io/master/en/index.html

closed 2020-05-27 by TSC21