PX4/PX4-SITL_gazebo-classic #501: Image Capture Not working in GAZEBO-PX4 SITL

HarshilNaikk, 2020-05-22:
> `harshil@harshil-Lenovo-ideapad-320-15IKB:~/Firmware$ roslaunch px4 mavros_posix_sitl.launch 
... logging to /home/harshil/.ros/log/db5b0b1e-9bea-11ea-b77f-b0fc362b6d6b/roslaunch-harshil-Lenovo-ideapad-320-15IKB-17913.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://harshil-Lenovo-ideapad-320-15IKB:33905/

SUMMARY
========

CLEAR PARAMETERS
 * /mavros/

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /mavros/cmd/use_comp_id_system_control: False
 * /mavros/conn/heartbeat_rate: 1.0
 * /mavros/conn/system_time_rate: 1.0
 * /mavros/conn/timeout: 10.0
 * /mavros/conn/timesync_rate: 10.0
 * /mavros/distance_sensor/hrlv_ez4_pub/field_of_view: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/frame_id: hrlv_ez4_sonar
 * /mavros/distance_sensor/hrlv_ez4_pub/id: 0
 * /mavros/distance_sensor/hrlv_ez4_pub/orientation: PITCH_270
 * /mavros/distance_sensor/hrlv_ez4_pub/send_tf: True
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/hrlv_ez4_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/laser_1_sub/id: 3
 * /mavros/distance_sensor/laser_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/laser_1_sub/subscriber: True
 * /mavros/distance_sensor/lidarlite_pub/field_of_view: 0.0
 * /mavros/distance_sensor/lidarlite_pub/frame_id: lidarlite_laser
 * /mavros/distance_sensor/lidarlite_pub/id: 1
 * /mavros/distance_sensor/lidarlite_pub/orientation: PITCH_270
 * /mavros/distance_sensor/lidarlite_pub/send_tf: True
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/x: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/y: 0.0
 * /mavros/distance_sensor/lidarlite_pub/sensor_position/z: -0.1
 * /mavros/distance_sensor/sonar_1_sub/id: 2
 * /mavros/distance_sensor/sonar_1_sub/orientation: PITCH_270
 * /mavros/distance_sensor/sonar_1_sub/subscriber: True
 * /mavros/fake_gps/eph: 2.0
 * /mavros/fake_gps/epv: 2.0
 * /mavros/fake_gps/fix_type: 3
 * /mavros/fake_gps/geo_origin/alt: 408.0
 * /mavros/fake_gps/geo_origin/lat: 47.3667
 * /mavros/fake_gps/geo_origin/lon: 8.55
 * /mavros/fake_gps/gps_rate: 5.0
 * /mavros/fake_gps/mocap_transform: True
 * /mavros/fake_gps/satellites_visible: 5
 * /mavros/fake_gps/tf/child_frame_id: fix
 * /mavros/fake_gps/tf/frame_id: map
 * /mavros/fake_gps/tf/listen: False
 * /mavros/fake_gps/tf/rate_limit: 10.0
 * /mavros/fake_gps/tf/send: False
 * /mavros/fake_gps/use_mocap: True
 * /mavros/fake_gps/use_vision: False
 * /mavros/fcu_protocol: v2.0
 * /mavros/fcu_url: udp://:14540@loca...
 * /mavros/gcs_url: 
 * /mavros/global_position/child_frame_id: base_link
 * /mavros/global_position/frame_id: map
 * /mavros/global_position/gps_uere: 1.0
 * /mavros/global_position/rot_covariance: 99999.0
 * /mavros/global_position/tf/child_frame_id: base_link
 * /mavros/global_position/tf/frame_id: map
 * /mavros/global_position/tf/global_frame_id: earth
 * /mavros/global_position/tf/send: False
[... 360 lines cut ...]
    return wp_list

if __name__ == '__main__':
	rospy.init_node('ardu_mavros_node', anonymous=True)
	mavutil.set_dialect("ardupilotmega")
	master = mavutil.mavlink_connection('udpin:0.0.0.0:14550')
	master.wait_heartbeat()

	setarm(1)
	time.sleep(2)
	takeoff(35)
	time.sleep(1)
	waypoint('/home/harshil/BSDC-Ardupilot_and_Mavros/ArduMavros/src/mavros_py/waypointfiles/wayptmav.txt')
	time.sleep(1)
	master.mav.command_long_send(
    master.target_system,
    master.target_component,
    mavutil.mavlink.MAV_CMD_IMAGE_START_CAPTURE,
    1,
    0, 3, 20, 0, 0, 0, 0)
	setmode('AUTO.MISSION')

`

@julianoes

julianoes, 2020-05-25:
It looks like you're using ArduMavros. I have no idea if that works with PX4 like that.

And please put ``` around pasted log output or source to make it more readable.

Jaeyoung-Lim, 2020-06-07:
@HarshilNaikk Tried with QGC and PX4 SITL and looks functional. Have you tried without using mavros?

appetito, 2020-07-30:
You can check Gazebo logs for details (~/.gazebo/server-11345/default.log on my machine)

Jaeyoung-Lim, 2020-07-31:
@appetito Thanks!

Closing since the original author of the issue has not responded

closed 2020-07-31 by Jaeyoung-Lim