PX4/PX4-SITL_gazebo-classic #503: When changing a world in Gazebo, the drone model is duplicated.

petergerten, 2020-05-26:
To reproduce:
```
1. make px4_sitl gazebo_typhoon_h480
2. Change something in the world
3. "save"
4. Start gazebo with the modified world file
```
Now there are too h480 on top of each other.
sitl_gazebo seems to instantiate the drone independently from the world file.
I noticed that I can manually delete the definition for the h480 model in the file and it works. 
But how would I for example change the pose of the model? (i.e. start from the balcony of a house)

petergerten, 2020-05-26:
Switching from Gazebo 9 to Gazebo 10 fixed this.

closed 2020-05-26 by petergerten

Jaeyoung-Lim, 2020-05-26:
@petergerten It is most likely that it is not related to the gazebo version, but gazebo server was still running when you restarted gazebo

petergerten, 2020-05-26:
@Jaeyoung-Lim thanks, I will keep that in mind.