PX4/PX4-SITL_gazebo-classic #510: Fail to add motor_failure_plugin on iris.sdf

hslyu, 2020-06-05:
Hi everyone.
I tried to utilize [motor_failure_plugin](https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_motor_failure_plugin.cpp) on default iris model, so I modified **sitl_gazebo/models/iris/iris.sdf**, which is generated by "**make px4_sitl gazebo**" command on the Firmware directory, which is cloned by [Firmware repo](https://github.com/PX4/Firmware).

I add 
`    <plugin name="motor_failure" filename="libgazebo_motor_failure_plugin.so">
        <robotNamespace/>
        <ROSMotorNumSubTopic>/motor_failure/motor_number</ROSMotorNumSubTopic>
        <MotorFailureNumPubTopic>/gazebo/motor_failure_num</MotorFailureNumPubTopic>
    </plugin>
`
in the iris.sdf file.
After I implement gazebo simulation, I got this error:
`SITL ARGS
sitl_bin: /root/src/Firmware/build/px4_sitl_default/bin/px4
debugger: none
program: gazebo
model: none
world: none
src_path: /root/src/Firmware
build_path: /root/src/Firmware/build/px4_sitl_default
empty model, setting iris as default
GAZEBO_PLUGIN_PATH :/root/src/Firmware/build/px4_sitl_default/build_gazebo
GAZEBO_MODEL_PATH :/root/src/Firmware/Tools/sitl_gazebo/models
LD_LIBRARY_PATH /root/catkin_ws/devel/lib:/opt/ros/melodic/lib:/usr/local/nvidia/lib:/usr/local/nvidia/lib64:/root/src/Firmware/build/px4_sitl_default/build_gazebo
empty world, setting empty.world as default
ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
AL lib: (EE) ALCplaybackAlsa_open: Could not open playback device 'default': No such file or directory
**gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::transport::Node; typename boost::detail::sp_member_access<T>::type = gazebo::transport::Node*]: Assertion `px != 0' failed.
/root/src/Firmware/Tools/sitl_run.sh: line 130: 18650 Aborted                 (core dumped) gzserver** "${src_path}/Tools/sitl_gazebo/worlds/empty.world"
SITL COMMAND: "/root/src/Firmware/build/px4_sitl_default/bin/px4" "/root/src/Firmware"/ROMFS/px4fmu_common -s etc/init.d-posix/rcS -t "/root/src/Firmware"/test_data
INFO  [px4] Creating symlink /root/src/Firmware/ROMFS/px4fmu_common -> /root/src/Firmware/build/px4_sitl_default/tmp/rootfs/etc
`
If you have any idea about add plugin on model, please let me know.
Do I need to modify the empty.world?
+ I executed the SITL gazebo simulation using 
`/root/src/Firmware/Tools/sitl_run.sh /root/src/Firmware/build/px4_sitl_default/bin/px4 none gazebo none none /root/src/Firmware /root/src/Firmware/build/px4_sitl_default`, which is the same as `make px4_sitl gazebo`

Jaeyoung-Lim, 2020-06-07:
@hslyu Can you run the simulation with the same command, but without the motor failure plugin? It is not the world that is failing. There is something that is being failed to initialize that is on a shared_ptr

hslyu, 2020-06-09:
@Jaeyoung-Lim Yes, without motor failure plugin, I can run simulation with the same command. Also, I can implement motor failure using cpp file which is compiled outside of the simulation source.

Since I'm not used to use the gazebo plugin, it may just be the .sdf format issue

Jaeyoung-Lim, 2020-06-09:
@hslyu I am not sure what you are trying to do. Are you saying that the default firmware SITL is broken?

Jaeyoung-Lim, 2020-10-05:
@hslyu Any updates?

Jaeyoung-Lim, 2020-11-25:
Closing since there were no updates from @hslyu

closed 2020-11-25 by Jaeyoung-Lim

mariam0amin, 2024-05-21:
@hslyu  did you manage to work this put???

hslyu, 2024-05-22:
@mariam0amin Yes, somehow. But the issue has been too outdated I don't know exactly how I made it :( 
https://youtu.be/UG6amcLl9cY