PX4/PX4-SITL_gazebo-classic #513: CI failing on MacOS (again)

Jaeyoung-Lim, 2020-06-07:
Seems like the profobuf version being used in Mac OS is different with the rest of linux releases. 

Error Log: https://travis-ci.org/github/PX4/sitl_gazebo/jobs/695729908
```
In file included from /Users/travis/build/PX4/sitl_gazebo/build/Wind.pb.h:35:
/usr/local/Cellar/gazebo9/9.13.0_1/include/gazebo-9/gazebo/msgs/vector3d.pb.h:52:51: error: no type named 'AuxillaryParseTableField' in namespace 'google::protobuf::internal'; did you mean 'AuxiliaryParseTableField'?
  static const ::PROTOBUF_NAMESPACE_ID::internal::AuxillaryParseTableField aux[]
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
/usr/local/include/google/protobuf/generated_message_table_driven.h:141:7: note: 'AuxiliaryParseTableField' declared here
union AuxiliaryParseTableField {
      ^
1 error generated.
make[2]: *** [CMakeFiles/physics_msgs.dir/Wind.pb.cc.o] Error 1
make[1]: *** [CMakeFiles/physics_msgs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

LorenzMeier, 2020-06-07:
OSRF needs to fix it. They need to re-build gazebo9. In the meantime building it locally from source fixes this.

```
brew reinstall --build-from-source gazebo9
```

TSC21, 2020-06-07:
@Jaeyoung-Lim the above seems like a real typo. Auxillary vs Auxiliary.

Jaeyoung-Lim, 2020-06-07:
@TSC21 I agree, but I don't think the typo is not coming from this repo.

Jaeyoung-Lim, 2020-06-09:
Related: https://github.com/osrf/homebrew-simulation/issues/1042

Jaeyoung-Lim, 2020-11-19:
Fixed by https://github.com/PX4/PX4-SITL_gazebo/pull/613

closed 2020-11-19 by Jaeyoung-Lim