PX4/PX4-SITL_gazebo-classic #518: Unable to connect to custom designed hexrotor

deepak-1530, 2020-06-14:
I have written the code for a custom hexrotor and have followed the typhoon_h480 model for the same. Inspite of adding the mavlink plugins in the SDF file I am not able to connect to the drone.

TSC21, 2020-06-14:
@deepak-1530 hi! What do you mean with not being able to connect the drone? Have you created the respective init.d initialization file in the Firmware side? Also, have you created a mixer for that configuration or are you using the same as the Typhoon?

deepak-1530, 2020-06-14:
When i do make px4_sitl gazebo, it shows

INFO  [px4_work_queue] creating: wq:hp_default, priority: 87, stack: 16384 bytes
INFO  [px4_work_queue] creating: wq:att_pos_ctrl, priority: 88, stack: 16384 bytes
INFO  [px4_work_queue] creating: wq:rate_ctrl, priority: 99, stack: 16384 bytes
INFO  [init] Mixer: etc/mixers/quad_w.main.mix on /dev/pwm_output0
INFO  [mavlink] mode: Normal, data rate: 4000000 B/s on udp port 14570 remote port 14550

After this, nothing happens. 
I have created the init.d posix initialization file. For the mixer, I have used the same as typhoon. It will be amazing if you can tell me how to create one.

TSC21, 2020-06-14:
> When i do make px4_sitl gazebo, it shows
>
> INFO [px4_work_queue] creating: wq:hp_default, priority: 87, stack: 16384 bytes
> INFO [px4_work_queue] creating: wq:att_pos_ctrl, priority: 88, stack: 16384 [...]

To start the configuration you want, `make px4_sitl gazebo` is not enough, since that will spawn the Iris model by default. Have you add your aircraft as a CMake target?

deepak-1530, 2020-06-14:
Yes I have done that as well. In the iris.world file, I have used my custom drone model instead of iris drone. I am getting my drone in the gazebo world but unable to connect to it over mavlink

deepak-1530, 2020-06-14:
[image]
I followed these steps.

Jaeyoung-Lim, 2020-10-05:
@deepak-1530 Any updates?

deepak-1530, 2020-10-05:
Hello @Jaeyoung-Lim I updated the SDF file of typhoon (with our own meshes generated from the SolidWorks model) and also changed the parameters in mixer file to get my own drone model as i couldn't get the original file to work.

Jaeyoung-Lim, 2020-10-05:
@deepak-1530 Sounds like you made it work in the end. Congratulations!

Can we close this issue in that case?

deepak-1530, 2020-10-05:
Yes. I'll close it

closed 2020-10-05 by deepak-1530