PX4/PX4-SITL_gazebo-classic #519: Pitching moment implementation

lum-supakorn, 2020-06-15:
Hello, all.

I'm looking to implement pitching moment support for the [LiftDragPlugin](http://gazebosim.org/tutorials?tut=aerodynamics&cat=physics) since its latest version only supports lift and drag calculation, and I need pitch control for my current dual rotor tail-sitter project.

After looking through the source code and the documentation, the plugin calculates lift by simplifying the Cl-⍺ into two linear relationships between Cl and ⍺: pre-stall angle and post-stall angle of attack.

It seems to me that this simplification cannot be applied to Cm-⍺ curve (and also Cd-⍺ curve) since both curves look vastly different than the Cl-⍺ curve.

My question is how should I go about implementing the pitching moment behavior given the differently looking curve. Would it be a good idea to implement some kind of linear interpolation lookup table (given a *-⍺ relationship user input file)?

Jaeyoung-Lim, 2020-06-16:
@lum-supakorn I have enabled setting cm-alpha from the sdf in https://github.com/PX4/sitl_gazebo/pull/516 and you are right that the oversimplification makes it hard to model cm-alpha. 

Do you think we can model it into a low polynomial? or even a first order function around small angle of attacks? which airfoil are you considering as reference?

lum-supakorn, 2020-06-16:
@Jaeyoung-Lim I went for the lookup table with linear interpolation in my local repository. It seems to give good real-time performance and accurate aerodynamic parameter since the lookup table is constructed directly from an XFLR5 polar file.

However, I faced an issue with the control surface deflection angle (more details here: https://aviation.stackexchange.com/questions/78908/is-there-any-non-experimental-way-to-obtain-cl-cd-and-cm-as-a-function-of-the [I ran the analysis with a simple rectangular with with the MH60 airfoil and 0% static margin])

For CL and CM, the curves seem linear enough so describing it with a slope should be adequate; however, the CD curve does not look linear and we might need some other method.

Furthermore, as the deflection angle increases or decreases from the neutral point, each curve behaves differently (CL and CM curve seems to shift linearly while CD curve translates in a somewhat parabolic manner). In Gazebo, the effect of the control surface deflection angle is only realized for lift. And it was implemented such that the CL curve is shifted _up or down_ depending on the deflection angle. This is not the case since CL curve moves from side to side as well (e.g. deflecting the control surface down also decreases the stall angle).

My current idea is to allow the plugin user to input a CL/CD/CM curve (exported directly from XFLR5) at two control surface deflection endpoints and linearly interpolate between these two curves to construct a three-dimensional lookup table where I can pick any CL/CD/CM value by specifying the angle of attack and the deflection angle.

Jaeyoung-Lim, 2020-06-16:
@lum-supakorn Thanks for looking into this. 

IMHO I would prefer we still enable a delta wing flyable without the exports so that it works in a minimal setup and then we support the more accurate CL/CD/CM curve exported from XFLR5. This way we can enable users that do not use XFLR5, but direct people who want more accurate liftdrag simulation to use the tool. 

I agree the assumptions are not accurate but it is sufficient to test the software. Note that gazebo is not designed to be capable of simulating accurate dynamics but are more focused on enabling software testing with various sensors / cameras etc. Therefore I am a bit careful on over-complicating the lift drag calculations.

lum-supakorn, 2020-06-16:
@Jaeyoung-Lim I agree. I’m afraid that I might have over-complicated things.

For the delta wing case, can we assume that the Cm-alpha curve is linear and that the elevon deflection angle only shifts the curve up or down?

Jaeyoung-Lim, 2020-06-16:
@lum-supakorn If it flies, and we can fly a mission with it I would say this is already a first good step

lum-supakorn, 2020-06-16:
@Jaeyoung-Lim Got it. Will look into this as soon as possible.

lum-supakorn, 2020-06-18:
> I agree the assumptions are not accurate but it is sufficient to test the software. Note that gazebo is not designed to be capable of simulating accurate dynamics but are more focused on enabling [...]

@Jaeyoung-Lim Sneaking in a little bit for a follow up question. If I want to get serious with the vehicle dynamics for my control algorithm testing, which platform should I be using instead of Gazebo?

Jaeyoung-Lim, 2020-06-18:
@lum-supakorn I would suggest that we need to move towards integrating into jsbsim directly.

lum-supakorn, 2020-06-19:
@Jaeyoung-Lim I would like to now move on to making the delta wing fly. Is the build target for delta wing already there? I can't seem to find it.

Jaeyoung-Lim, 2020-06-19:
@lum-supakorn Thanks! You can continue this branch https://github.com/PX4/sitl_gazebo/tree/pr-deltawing

Jaeyoung-Lim, 2020-08-12:
@lum-supakorn I believe it is looking good on simulating aerodynamic pitch moments in gazebo.

There still needs to be improvement on simulating the liftdrag elements, but closing

closed 2020-08-12 by Jaeyoung-Lim

lum-supakorn, 2020-08-12:
@Jaeyoung-Lim Thank you. Sorry, I've been busy working on my senior project.