PX4/PX4-SITL_gazebo-classic #520: Motor model does not take wind into account correctly

ThomasRigi, 2020-06-16:
Summarising the conversation with @Jaeyoung-Lim  on Slack:

I have observed that the airspeed tracking in the presence of wind was much worse than without and that the thrust scaled with the groundspeed and not the airspeed. The following two logs show this.

Plane: https://logs.px4.io/plot_app?log=71bd83d8-52a8-42e4-bc32-3b28cdc88e6f
Standard VTOL: https://logs.px4.io/plot_app?log=0a4671dd-2258-4849-bded-e475d6818173

It is very remarkable with the VTOL which is commanded to fly at 22m/s airspeed. Both logs have 5m/s wind blowing from the south. With tailwind, it can't get to the desired airspeed even with full thrust. With headwind, 35% thrust are enough to reach the setpoint.

I first suspected that the drag of the vehicle was wrongly scaled with groundspeed instead of airspeed in https://github.com/PX4/sitl_gazebo/blob/master/src/liftdrag_plugin/liftdrag_plugin.cpp. In the meantime I verified that this part works correctly.

@Jaeyoung-Lim has directed me to the rotor drag calculations. I have identified https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_motor_model.cpp#L208 as the main culprit. The force that the motor develops is scaled based on the body velocity and not the relative wind velocity, which is only calculated further below. On top of that, the comment states that it is zero at 50m/s, but it actually already is zero at 25m/s. (Which explains well why the VTOL was struggling so badly.) 

I don't fully understand the motor model plugin, but the thrust should certainly be scaled on airspeed and not groundspeed. But I'm not sure if it should also take into account the motor orientation here? 

Also, I think it would be good to increase the speed at which scalar becomes zero to enable faster flights with FW aircrafts. What do you think? Maybe it could be set as a parameter in the sdf file?

Jaeyoung-Lim, 2020-07-05:
@ThomasRigi Thanks for reporting this. Were you able to have a look on how to fix this?

ThomasRigi, 2020-07-06:
@Jaeyoung-Lim No, not really. Maybe it's enough to move https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_motor_model.cpp#L222 above L207 and adapt L207 and L208 accordingly:

`ignition::math::Vector3d relative_wind_velocity = body_velocity - wind_vel_;`
`double vel = relative_wind_velocity.Length();`
`double scalar = 1 - vel / 25.0; // at 50 m/s the rotor will not produce any force anymore`

But I currently don't have time to test it. And I don't really understand the physics behind this `double scalar`, so I'm not sure if it is the correct way

notoriou5, 2020-07-10:
@ThomasRigi maybe I can help a bit to clear things up here. If you have a look at this [diagram](https://aviation.stackexchange.com/questions/10031/is-there-a-special-way-to-calculate-the-parasitic-drag-produced-by-the-propell) you see that Thrust is decreasing the faster the propeller gets blown at (perpendicular to the rotor plane). The above representation of `double scalar` is a highly linear model of this.

Maybe you are familiar with the advance ratio (J = V/(n*D)). I think this is a much better way to represent the loss of thrust due to the inflow (although the advance ratio changes for each propeller-motor setup)

Thats how I tried to change it: 

 ```
  ignition::math::Vector3d velocity_perpendiculat_to_rotor = (body_velocity * joint_axis) * joint_axis;
  ignition::math::Vector3d velocity_in_rotor_plane = body_velocity - (body_velocity * joint_axis) * joint_axis;

  double rotor_inflow_velocity = sqrt(velocity_perpendiculat_to_rotor.Dot(velocity_perpendiculat_to_rotor));
  double rounds_per_second = abs(motor_rot_vel_ * rotor_velocity_slowdown_sim_);
  rounds_per_second = ignition::math::clamp(rounds_per_second, 0.0, max_rot_velocity_);

  // Scale down thrust linearly with advance ratio (does not match reality completely!)
  // If maximal advance ratio is reached, the rotors will produce no thrust anymore. 
  // The maximal advance ratio depens on the pitch of the propeller blade
  // @see: https://nptel.ac.in/content/storage2/courses/101106041/chapter%204%20Lecture%2014%2020-12-2011.pdf
  double advance_ratio = ignition::math::fixnan(rotor_inflow_velocity / (rounds_per_second * rotor_diameter_));
  advance_ratio = ignition::math::clamp(advance_ratio, 0.0, 2.0);
  double scale = 1 - advance_ratio / advance_ratio_limit_;
  scale = ignition::math::clamp(scale, 0.0, 1.0);
```

But it is still linear due to the fact that my decay is not curved and needs further improvement. 

Maybe it helps you out a bit,

ThomasRigi, 2020-07-10:
@notoriou5 Thank you for your explications. It's much more clear now (though still not completely :sweat_smile:) - I assume that the rotational axis of a revolute joint is always the z-axis, independent of the orientation of this joint wrt the rest of the aircraft? otherwise `link_->AddRelativeForce(ignition::math::Vector3d(0, 0, force * scalar));` does not make sense to me.

For the time being I can live with the simple linear approximation (Don't get me wrong - it's really cool if you improve it :wink:  it's just not (yet) very important for our simulations as we typically only fly at one airspeed and once we have the thrust correct at that operating point it's "good enough" for now.), but according to my understanding it should only take into account the airflow perpendicular to the rotor plane. Is this right? So it should at least take into account this:
`ignition::math::Vector3d relative_wind_velocity = body_velocity - wind_vel_;`
`ignition::math::Vector3d velocity_perpendicular_to_rotor = (relative_wind_velocity * joint_axis) * joint_axis;`
`double vel = body_velocity_perpendicular.Length();`
`double scalar = 1 - vel / 25.0;`

Do you agree?

notoriou5, 2020-07-10:
> @notoriou5 Thank you for your explications. It's much more clear now (though still not completely ) - I assume that the rotational axis of a revolute joint is always the z-axis, independent of the orientation of this joint wrt the rest of the aircraft? otherwise `link_->AddRelativeForce(ignition::math::Vector3d(0, 0, force * scalar));` does not make sense to me.
> 
Exactly, thats how I understood it as well (I think I also tested it with some debug values but @Jaeyoung-Lim please correct me if I am wrong)

> For the time being I can live with the simple linear approximation (Don't get me wrong - it's really cool if you improve it  it's just not (yet) very important for our simulations as we typically only fly at one airspeed and once we have the thrust correct at that operating point it's "good enough" for now.), but according to my understanding it should only take into account the airflow perpendicular to the rotor plane. Is this right? So it should at least take into account this:
> `ignition::math::Vector3d relative_wind_velocity = body_velocity - wind_vel_;`
> `ignition::math::Vector3d velocity_perpendicular_to_rotor = (relative_wind_velocity * joint_axis) * joint_axis;`
> `double vel = body_velocity_perpendicular.Length();`
> `double scalar = 1 - vel / 25.0;`
> 
> Do you agree?

Shouldnt it be`double vel = velocity_perpendicular_to_rotor.Length();` ? In case the rotor and the body have the same orientation it doesn't matter though.

ThomasRigi, 2020-07-10:
> Shouldnt it bedouble vel = velocity_perpendicular_to_rotor.Length(); ? In case the rotor and the body have the same orientation it doesn't matter though.

Bad copy-pasting, sorry. You are of course completely right. I edited my comment above with the corrected version.

ThomasRigi, 2020-07-22:
Today I tried to implement the above outlined fix but it didn't work. (I tested on the v1.11.0-rc1 commit since I can't launch the windy world on master (#551))

I tried to understand what was going on and what I need to change to get it working correctly, but after some hours I ran out of ideas and abandoned for now. Unfortunately I forgot to save the logs before cleaning the directory :/ but basically the same problem still persists, thrust is still not based on airspeed and varies a lot depending on the direction you're flying relative to the wind.

Jaeyoung-Lim, 2020-08-04:
@ThomasRigi I have made a fixed the issue in https://github.com/PX4/sitl_gazebo/pull/552 and was a regression with one of the PRs merged recently.

Jaeyoung-Lim, 2020-08-12:
@ThomasRigi Did you manage to look into this issue?

ThomasRigi, 2020-08-12:
@Jaeyoung-Lim not since my last comment above. I tried the fixes I outlined before, but they didn't work and now I have run out of ideas of how to fix the bug. :/

Jaeyoung-Lim, 2020-08-12:
@ThomasRigi Do you mean that you still have the bug where you can't run the windy world?

ThomasRigi, 2020-08-12:
@Jaeyoung-Lim No I mean the bug about the thrust being dependent on groundspeed rather than airspeed.

I can now run the windy world, but as I mentioned in https://github.com/PX4/sitl_gazebo/pull/557 the wind direction is somehow not correct. I tried to look into that for about two hours but to no avail as I couldn't really make sense of the coordinate frame rotations  :/

Jaeyoung-Lim, 2020-08-12:
@ThomasRigi Okay, it would be great if you could share your findings from your investigation

ThomasRigi, 2020-08-12:
@Jaeyoung-Lim Unfortunately I forgot to save the logs somewhere permanent :(  Basically I tried to modify `double vel` just before `double scalar = 1 - vel / 25.0; // at 50 m/s the rotor will not produce any force anymore` in different ways to try to take into account the wind. None of them worked, though.

The first try was the one outlined in https://github.com/PX4/sitl_gazebo/issues/520#issuecomment-654185564. Then I tried https://github.com/PX4/sitl_gazebo/issues/520#issuecomment-656619886 and some modifications of it, but none of them gave reasonable results.

closed 2020-12-17 by Jaeyoung-Lim

Jaeyoung-Lim, 2020-12-17:
@ThomasRigi @notoriou5 I have fixed the motor thrust issue in #674

Would be great if you can try it out and provide feedback. Thanks!

ThomasRigi, 2020-12-18:
Many thanks @Jaeyoung-Lim ! Works as expected now. :+1:

Jaeyoung-Lim, 2020-12-18:
@ThomasRigi Thanks for coming back, glad that it fixed it