PX4/PX4-SITL_gazebo-classic #526: Iris drone parameters

Black-Ali, 2020-06-19:
Hello!

I was trying to use the iris quadrotor parameters found in iris.xacro and I have few questions about it:
1- The motor constant is given for a rotor speed given in rpm aw rad/sec? I am asking since the unit is Kg.m/s^2 (i.e. N) without specifying if N/rpm^2 or N/(rad/s)^2
2- Is it normal that "arm_length_back_y" & "arm_length_front_y" are different even that thos of x are equal?
3- The moment constant (=0.06) is the constant to calculate the reactive troque of a rotor? (i.e. T=(moment constant)*(rotor's speed in rad/s)^2) 

I appreciate any answer or help ! :)

Jaeyoung-Lim, 2020-07-05:
@Black-Ali You can look at how the motor constant / moment constant is being used in the `gazebo_motor_model` : https://github.com/PX4/sitl_gazebo/blob/master/src/gazebo_motor_model.cpp

2. That is simply the realistic values relative to the CoM for the iris vehicle

Black-Ali, 2020-07-05:
Thank you a lot for your reply! 
I'll check it

closed 2020-08-12 by Jaeyoung-Lim