PX4/PX4-SITL_gazebo-classic #531: LiftDragPlugin: What's the "moment" term in total force expression?

lum-supakorn, 2020-06-22:
What's the reason behind putting the [moment term in the total force expression?](https://github.com/PX4/sitl_gazebo/blob/309f7b0ace2d6d0afea59fe5edf3356b26d996c9/src/liftdrag_plugin/liftdrag_plugin.cpp#L397) It was initially commented out along with other [moment-related terms](https://github.com/PX4/sitl_gazebo/blob/309f7b0ace2d6d0afea59fe5edf3356b26d996c9/src/liftdrag_plugin/liftdrag_plugin.cpp#L399), but I uncommented it when I implemented the pitching moment calculation. Looking back at the code, the moments generated from lift and drag make sense but what's up with the force generated by moment?

Jaeyoung-Lim, 2020-06-22:
@lum-supakorn I think you are right, sorry I didn't see that in the review. Lets remove it

lum-supakorn, 2020-06-22:
@Jaeyoung-Lim Do I open a new PR, or is there a better way to remove this small piece of code?

Jaeyoung-Lim, 2020-06-22:
Please open a new PR

Jaeyoung-Lim, 2020-06-23:
Fixed in https://github.com/PX4/sitl_gazebo/pull/532

closed 2020-06-23 by Jaeyoung-Lim

lum-supakorn, 2020-06-29:
@Jaeyoung-Lim Do you know if `AddForceAtRelativePosition` automatically adds moment to the body? If so, `lift.Cross(momentArm)` and `drag.Cross(momentArm)` wouldn't be necessary.

Jaeyoung-Lim, 2020-06-29:
@lum-supakorn Good point, I think you are right. Wondering how this was affecting the dynamics until now

lum-supakorn, 2020-06-29:
@Jaeyoung-Lim So `AddForceAtRelativePosition` automatically adds moment to the link? That's surprising.

Jaeyoung-Lim, 2020-06-29:
@lum-supakorn would you be able to verify this? It will generate moments around the CoM

lum-supakorn, 2020-06-29:
@Jaeyoung-Lim The `Apply Force/Torque` function in Gazebo seems to automatically resolve the moment if a force is applied. This can be tested by applying force from this function away from the CG and watching a model rotate. I'll write a custom model plugin to verify if the `AddForceAtRelativePosition` function does the same thing.

lum-supakorn, 2020-07-02:
I did not write a new plugin, but upon inspecting Open Dynamics Engine's source code, I found that `dBodyAddForceAtRelPos` which is called by `ODELink::AddForceAtRelativePosition` indeed [resolves the moment automatically.](https://github.com/osrf/gazebo/blob/7200818c3f1b905505d25c66d5ad004cd0003e88/deps/opende/src/ode.cpp#L633) This means that we do not need to manually cross the lift and drag vectors and add the torque.

Jaeyoung-Lim, 2020-07-02:
@lum-supakorn Awesome, that makes sense

Jaeyoung-Lim, 2020-07-09:
@lum-supakorn Any updates regarding the deltawing model?

lum-supakorn, 2020-07-09:
@Jaeyoung-Lim Sorry but there are no updates yet. I've been trying to solve the differential thrust yaw stability problem for the dual-rotor tailsitter model. I thought the delta wing might face the same issue but noticed it already has rudders. Also, I need to port PX4 test suite to the latest ROS version so that I can verify that pitching moment control works. Will get back to this as soon as I get all those done.

P.S. I saw you in PX4 Dev Summit!